Screenly/Anthias #837: API call for navigate to any asset error

wwj718, 2018-06-22:
#### Overview of the Issue
[image]

### Environment

* Raspberry Pi 3B
* Screenly OSE Version: commit 62bfe2ff477122ee97bf1b1017e94abffad52013 (Thu May 3 14:53:18 2018)

antonmolodykh, 2018-06-22:
@wwj718  I think it's in `&` symbol. try something like `curl -s "http://127.0.0.1:8080/api/v1/assets/control/asset&64c9830eae4e484db276e1c6259d2849"`

vpetersson, 2018-06-22:
Yep. That's not a bug, `&` in Bash has means run the next command (vastly simplified). @antonmolodykh solution above should solve it.

closed 2018-06-22 by vpetersson

wwj718, 2018-06-22:
@antonmolodykh @vpetersson Thanks a lot !  it works!