Screenly/Anthias #842: Perform facelift

vpetersson, 2018-06-22:
The look and feel in OSE is starting to look a bit dated. After we have upgraded Bootstrap in #840, let's switch to a Bootstrap theme. Here's [one candidate](https://hackerthemes.com/bootstrap-themes/growth/).

antonmolodykh, 2018-08-15:
I applied this theme, but the visible changes are very small. I think OSE needs a more serious redesign
[image]
[image]

madsrh, 2018-11-27:
This really didn't work out 😆 but since I spent the time working on it, I might as well share it here for conversation.

[image]

> ...but I think that we need a more complex redesign than a simple theme change.

If I can, I would love to give this a try with a few mockups

rusko124, 2018-11-27:
@madsrh It looks very good, I like it!
@vpetersson What do you think?

vpetersson, 2018-11-27:
@madsrh Thanks! I'll loop in our staff designer to get his feedback too.

L4cko, 2018-11-30:
> @madsrh Looks good! We just need to fix the colors.

rusko124, 2019-04-19:
@vpetersson I would like to see this in Sprint 10

closed 2019-09-10 by vpetersson