Screenly/Anthias #843: Switch to Balena

vpetersson, 2018-06-22:
The experimental branch is currently using Docker. We should switch to using [Balena](https://www.balena.io/) instead as it is written for this particular use case.

antonmolodykh, 2018-07-10:
At the moment `balena pull` works much slower than `docker pull`. I opened a [ticket](https://github.com/resin-os/balena/issues/86) about it.

antonmolodykh, 2018-08-25:
In fact, I changed the installation to use Balena, but when testing I could not run Balena on Pi1 and Pi Zero W. I asked about it in the [resin chat](https://gitter.im/resin-io/chat), but I have not received an answer yet.
The current progress on this task is [here](https://github.com/greatfruitomsk/screenly-ose/tree/Switch-to-Balena).

In addition, if we will use Balena, then using `command` module for ansible is not a good idea. It might make sense to write a `balena` module for ansible (something like `docker_container` but for Balena)

vpetersson, 2019-04-05:
@ealmonte32 you wanna take a stab at this one?

ealmonte32, 2019-04-05:
@vpetersson 
[image]

I honestly do want to learn docker(moby)/balena/containers/IoT, etc.., so I have been trying during my free time by messing with different Dockerfile configurations... so I make no promises. =]

vpetersson, 2019-04-08:
All good, @ealmonte32. Plenty of things to do here once you get to it ;)

closed 2019-06-21 by stale[bot]