Screenly/Anthias #848: ScreenlyWebView not dismissing alert dialog between assets

abrenoch, 2018-07-05:
It appears in the latest beta build if an `alert` dialog is show on a website, it will persist until the device has restarted.

To reproduce simply add a website that shows an `alert` dialog.

 I discovered this trying to debug a locally hosted website, and was not able to get it to go away without needing to reboot (tried to kill the web view process and things began acting up).

I'm running the 'My Michelle' release on a pi 3 b+

rusko124, 2018-09-11:
@vpetersson 
I confirm that this problem exists, i tried find solves for this problem but at the moment I can not offer a correct and beautiful solution. As far as I know, Screenly is planning to aborted the uzbl browser, so does it make sense now to look for a solution?

vpetersson, 2018-09-11:
@rusko124 

> As far as I know, Screenly is planning to aborted the uzbl browser, so does it make sense now to look for a solution?

Yes, this is the planned replacement for `uzbl` so it makes sense to spend time on finding a solution here.

vpetersson, 2018-09-13:
@brandfocus It's already in the works. You can try it out by switching to the experimental branch.

brandfocus, 2018-09-14:
@vpetersson There seem to be compile issues with Resin on the experimental branch...

`[31m[Error][39m        Error: The command '/bin/sh -c curl -s https://bootstrap.pypa.io/get-pip.py | python &&     pip install --upgrade -r /tmp/requirements.viewer.txt &&     pip install --upgrade -r /tmp/requirements.server.txt &&     pip install --upgrade -r /tmp/requirements.websocket_server.txt' returned a non-zero code: 1
[31m[Error][39m    Not deploying release.`

vpetersson, 2018-09-14:
Interesting. I don't think we tested the experimental release on Resin yet.

vpetersson, 2018-10-22:
Should be fixed. Please confirm @brandfocus.

closed 2018-10-22 by vpetersson