Screenly/Anthias #851: Audio support with a URL asset  (i.e. web page)?

Ne0nx3r0, 2018-07-07:
Is there support for playing sounds from web page / url assets in Screenly OSE?

I'm using HDMI audio and I've noticed when adding an MP4 asset audio works fine but when I try to play sounds from a URL / web site asset using the following there's no sound:

```
    function playAudio(fileUrl){
        var audio = new Audio(siteBase+'/audio/'+fileUrl);
        audio.play();
    }
```

When I view the URL on my desktop computer however it works as expected.

Is there a trick to getting audio to work from URLs or a supported method of playing sounds?

Raspberry Pi: BCM2835 - ARMv7 Processor rev 4 (v7l)
OS: Raspbian GNU/Linux 9 (stretch) via Screenly OSE image

vpetersson, 2018-07-09:
@Ne0nx3r0 Please try this in the experimental branch and see if it works there as we are moving to that browser.

vpetersson, 2018-07-20:
Sure. `git branch` from within /home/pi/screenly should tell you.