Screenly/Anthias #853: Asset set to "on" remains in inactive section

daecks, 2018-07-09:
Similar to issue #667 , our Screenly instances will after a while mark an "active" asset as "inactive", while the toggle switch is still set to "on".  Toggling the asset to "off" and then "on" will not put the asset back into the "active" section.  The asset needs to be removed/recreated for the "active/inactive" functionality to work.  Unlike as stated in #667, a reboot will *not* solve this.

Issue occurs randomly, however it will normally have appeared after a couple of days of operation.    As other issues suggested, I made sure the time on the device was correct.  See attached files for dmesg, journal and date logs/output.

Steps to reproduce: 
* Add an asset -> asset is displayed on TV.
* Wait for a few days.
* Asset is in inactive section, but still toggled "on".
* Toggle asset "off" then "on", asset remains in inactive section.  Asset needs to be recreated to work again.

### Environment

* Raspberry Pi Hardware Version: BCM2835, Revision a21041 (Pi 2 Model B)
* Screenly OSE Version: Running an SD card image downloaded from website.  commit 62bfe2ff477122ee97bf1b1017e94abffad52013

[screenly_journalctl.txt](https://github.com/Screenly/screenly-ose/files/2177160/screenly_journalctl.txt)
[screenly_date.txt](https://github.com/Screenly/screenly-ose/files/2177161/screenly_date.txt)
[screenly_dmesg.txt](https://github.com/Screenly/screenly-ose/files/2177162/screenly_dmesg.txt)

antonmolodykh, 2018-07-10:
If the device has no problems with time, an asset should be active if:
* user did not click "off" in the right column
* asset has not expired (see "start" and "end" columns).

can you confirm that both conditions are met?

daecks, 2018-07-10:
Gah!  That was it - the asset had expired.  I was ignorant of those fields, and was thinking if I just created an asset and did nothing else, why would it expire?  I've set it to 2099 - I think that's long enough!  Thanks.

closed 2018-07-10 by daecks