Screenly/Anthias #860: Delete old, unused assets

mscreations, 2018-07-12:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Client had an issue uploading a video to their screenly device due to unicode in the metadata (See #859 for the fix). Although the asset never finished being added (as it errors out in the middle), the file is still stored in ~/screenly_assets. It just becomes orphaned since the database no longer refers to it. Since the video file is around 38MB and the client tried at least 6 times, it's using up some storage with no client visible way to fix the wasted space. 

#### Reproduction Steps

* Try to add a video with unicode in metadata (prior to fix #859). When the scripts try to get the video duration, they will fail due to the unicode issue. The asset will not be added, but the file is still stored in ~/screenly_assets.

### Environment

* Raspberry Pi Hardware Version: Zero W
* Screenly OSE Version: e5f83ee3efeb85f1f9e7a812f3dbb31525082b2a roughly. The client has some very specific modifications that were applied on top of that particular commit. None of the modifications affects this issue. It still occurs with the vanilla commit. The modifications just add an internal clock display with customization for the client.

### Suggestion

I'd suggest a startup script that could check the database and any stored assets that are not referenced from the database would get deleted. A possibility would be to make it so you could trigger the script from settings.

antonmolodykh, 2018-07-12:
Screenly has a cron task:
```
$ crontab -l
#Ansible: Cleanup screenly_assets
0 1 * * * /usr/local/bin/screenly_utils.sh cleanup
```
It should delete a files after an unsuccessful uploading.

So I think you fixed the only possible problem why an unused file might not be deleted.

vpetersson, 2018-07-12:
@antonmolodykh how are we logging that? Perhaps the cleanup is broken.

antonmolodykh, 2018-07-12:
@vpetersson  We recently had a [ticket](https://github.com/Screenly/screenly-ose/issues/808) about it. The script really works.
The fact that the script has not cleared unused files as described in this ticket is the expected behavior (since in fact files were uploaded correctly)

vpetersson, 2018-07-12:
Yes, I recall. I'm not saying that it doesn't work in most cases. This is more for in case there is a bug somewhere.

mscreations, 2018-07-12:
I'll take a look when i get a chance. The device is never on at that time, so maybe might need to change it to run at boot. They turn it on and off with the tv during their skating sessions. That could be why they are being retained.

vpetersson, 2018-07-12:
@mscreations is the Raspberry Pi powered by the TV? If so, I suggest you change that ASAP as that will cause the Screenly installation to break. Replace that with a proper power supply.

mscreations, 2018-07-12:
It has a safe shutdown script that runs when power is shut off. Basically it is powered separately, but also has a connection to the tv. If tv is on power is turned on. When the tv shuts off, a script executes a safe shutdown eg. ``` sudo halt ```. I use something similar with my retro pi installations.

mscreations, 2018-07-19:
I had a chance to look at this today. There are a couple of duplicate files (based on the md5sum of the file) where only one appears in the database. The duplicates can be safely deleted. The cron job wouldn't help as it is only looking for *.tmp files, but the files that are in my assets directory are the final filename already. Some of the files are duplicates, but not 100% identical. As an example, files 13243677155e4223b3cdcd21d147202e and 5a728d885e8e478da6a847f56d2f1f55 differ by maybe 20 bytes (they are 26M files).

Directory structure of ~/screenly_assets:
[image]

Screenly.db assets table:
[image]

vpetersson, 2018-07-19:
Also, it should be pointed out that there is no de-duplication process in Screenly OSE. If you upload the same file 10 times, it will appear 10 times.

closed 2018-09-24 by stale[bot]