Screenly/Anthias #861: Datepicker/Timepicker not appearing

mscreations, 2018-07-12:
#### Overview of the Issue

After the upgrade to the newer version of Bootstrap, the Datepicker and Timepicker in the Asset edit view are not appearing properly. The time picker sometimes comes up, but not always. The date picker apparently is coming up, but is behind the dialog and hence does not work. Also, I am unable to type in a date as it keeps changing before you can finish typing in the correct date.

Also, is there anyway to make it so when one of the dialogs is up that it is scrollable? One of my clients uses his iPad and iPhone to access the configuration, and on the iPad it is only *just* sized enough to save changes. On the iPhone, you cannot see the save button to make and save changes. 

#### Reproduction Steps

1. ``` git checkout master ```
2. Click edit button on asset.
3. Click on Date to pull up the date picker. The date picker does not appear. On a larger monitor, you can see that it is appearing, but is behind the dialog rendering it useless.

### Environment

* Raspberry Pi Hardware Version: Zero W
* Screenly OSE Version: 3c543379dbc509c104c58e58a8cad95e7dfabf8d (master on screenly), 817f1d4e835a7ac02c14ed9897f41f57067578c2 (my repo with unicode patch)

antonmolodykh, 2018-07-12:
I see two possible causes of this problem:
1. if you did not restart your device after the update (or branch change)
2. If you have not cleared your browser's cache

In addition, I want to note that we recommend that you run the update, and not just switch the branch.

please let me know whether it helps you

mscreations, 2018-07-12:
All devices were restarted. One if the iPad's exhibited this issue when it was the first time i ever connected to the Screenly device. I also made sure to clear the cache in my browser. Running the update doesn't fix the issue. Hopefully I'll have a chance later to look at the issue in more depth.

vpetersson, 2018-07-12:
@antonmolodykh do we version (or use some other kind of cache burst) on static files?

antonmolodykh, 2018-07-12:
no, as far as I know we don't use such things

vpetersson, 2018-08-10:
Ok looking closer at this, I am pretty sure it's related to that. Let's add a cache buster.

closed 2018-09-12 by vpetersson