Screenly/Anthias #864: Wrong screen resolution on Westinghouse TV

arothmanmusic, 2018-07-16:
#### Overview of the Issue

I've set up Screenly OSE on a Raspberry Pi 3. It loads a single asset - a webpage with a full-screen 1920x1080 slideshow. This works perfectly on two Vizio TVs I've tested it with. However, the TV it's supposed to play on is a Westinghouse. On that particular TV, Screenly boots up 480p only. I get black bars down the sides and only the top left corner of the picture. I'm unable to get to the terminal with CTRL+ALT=F1 either.

### Environment

* Raspberry Pi Hardware Version: 3
* Screenly OSE Version: Latest (installed it this weekend)

vpetersson, 2018-07-24:
@arothmanmusic That's strange.

When you say "latest" does that mean the latest disk image?

> I'm unable to get to the terminal with CTRL+ALT=F1 on any of the TVs.

That should be ctrl+alt+f1

arothmanmusic, 2018-07-24:
Yeah, I meant ctrl+alt+f1 - just a typo. :)

"Latest" meant I went into the shell and ran $ bash <(curl -sL https://www.screenly.io/install-ose.sh) that day. I'm not sure what the version number was.

I actually have since overwritten the card and moved on to other methods because I couldn't get it to work within my deadline, so I don't have the details. Sorry about that...

vpetersson, 2018-08-13:
Closing out due to lack of response.

closed 2018-08-13 by vpetersson