Screenly/Anthias #865: Latest build broke streaming

quyet-nguyen, 2018-07-17:
Building screenly through resin.io or https://www.screenly.io/install-ose.sh broke the rtmp stream on my OBS (Both NvEVC and x264 ) > nginx + rtmp setup while the prebuilt image works.
Error log:
```
17.07.18 12:49:14 (+0700) Jul 17 05:49:14 8196940 python[224]: Showing asset rtmp://192.168.1.10/live/test (streaming)
17.07.18 12:49:14 (+0700) Jul 17 05:49:14 8196940 python[224]: current url is file:///tmp/screenly_html/black_page.html
17.07.18 12:49:19 (+0700) Jul 17 05:49:19 8196940 python[224]: Resource URI is not correct, remote host is not responding or request was rejected.
```

antonmolodykh, 2018-07-18:
Yes, we know that now there are problems with Screenly OSE on resin.io. now we recommend using the last official image.
what do you mean? 
> `or https://www.screenly.io/install-ose.sh`
You mean you ran a clean installation and ran into problems?

quyet-nguyen, 2018-07-18:
Both clean install and building through resin. I figured out the problem was because the gọi memory is by default 16mb, which is not enough for stream playback. At least 32mb required

closed 2018-07-18 by vpetersson