Screenly/Anthias #866: Wordpress web server heavy usage

dragonunit, 2018-07-18:
I have digital signage running on wordpress with Divi Builder pulling a couple of pages from that website. The web server is constantly getting hammered to the point where the external IP of the locations where Screenly is deployed get blocked. 

Is there any way to have screenly cache the pages for a week (they are only updated once weekly)

* Raspberry Pi Hardware Version: Raspi3 B+
* Screenly OSE Version: Paradise City

vpetersson, 2018-07-18:
No, you can't cache an entire website like that in Screenly. That said, something is probably wrong with the WordPress configuration if that's the case (unless you have thousands of devices hammering it). Try installing a caching plugin on Wordpress to see if that solves the issue.

dragonunit, 2018-07-18:
Its only 3 devices. I will check with the web host on what the issue is. I have caching plugins, minify and css combination. Weird.

vpetersson, 2018-07-19:
Interesting. Well, if three devices can take down your web server, something is definitely wrong.

vpetersson, 2018-08-10:
Closing out this as I'm pretty sure it's unrelated to Screenly.

closed 2018-08-10 by vpetersson