Screenly/Anthias #867: accessing terminal screen via keystroke

RoseLD, 2018-07-18:
Ctrl-Alt-F1 is not working to access the terminal screen for my Raspberry Pi running Screenly.
I'm using 2 different Raspberry Pi's to run Screenly.  I installed via NOOBS.  One I booted w/ a wired connection upon setup and installed Screenly so that I can boot up via Raspbian or Screenly.  The other I installed Raspbian only (did not have an internet connection upon installing), so I used the curl command to install Screenly.  I opted for the development version for this Pi.  I, therefore, can no longer access the terminal using Ctrl-Alt-F1 --- and only on this Pi.  Also, the Wi-Fi setup is behaving differently for the 2 Pi's.  
I recently updated via the curl command 
bash <(curl -sL https://www.screenly.io/install-ose.sh)
In order to get the Access Point to show up in between different WiFi's I have to delete .screenly.initialized via Putty (or, optimally, using a terminal screen on the Pi itself!!)

I can Putty into the Pi, but I'd really like the more expedient method of accessing the Pi via the keyboard.  I'm trying to set it up at a university with additional security to their Wi-Fi.  In order to use Putty, the laptop has to be on the same network.  I want the Pi in the gym & the laptop in the Pool area, so they won't be on the same WiFi network.   Wired connections are not an option in these two areas of the university.

THANKS FOR YOUR HELP!!

vpetersson, 2018-07-27:
@antonmolodykh could you please take a look?

rusko124, 2018-10-05:
Hi @RoseLD! I tried to completely reproduce your problem, but I did not succeed. Do you still have this problem? And if so, could you try installing again with the latest screenly updates?

qbicdesign, 2018-11-08:
put a video playing then try to reproduce. pretty sure if you're using only URL assets then it works.