Screenly/Anthias #868: Issues adding video to resin device

vpetersson, 2018-07-19:
#### Overview of the Issue

Issue with video parsing on Resin.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Deploy master to resin
* Press Add Asset
* Upload a video

You will receive the following error:

```
Server Error: RAN: '/usr/bin/omxplayer /data/screenly_assets/a3e9f0559ce041aea17bdddfe991dde3 --info' STDOUT: /usr/bin/omxplayer.bin: error while loading shared libraries: libWFC.so: cannot open shared object file: No such file or directory STDERR:
```

### Environment

* Raspberry Pi Hardware Version: Raspberry Pi Zero
* Screenly OSE Version: latest 379c417

vpetersson, 2018-07-19:
I tried to bump up the GPU_MEM to 96MB but that didn't make a difference.

rusko124, 2018-09-05:
I reproduced this issue with RaspberryPi Zero W.
This alert was solved after 
```
apt-get install libraspberrypi0
```
but still after next trying, I got a new alert
```
Server Error: RAN: '/usr/bin/omxplayer /data/screenly_assets/bb004e829dec47b8965966b877c15ad3 --info' STDOUT: * failed to open vchiq instance STDERR:
```
It is solved with  
```
chmod 777 /dev/vchiq
```
that it can be done more correctly

vpetersson, 2018-09-07:
@rusko124 also see [this comment](https://support.screenly.io/hc/en-us/community/posts/360016020073-Video-not-playing-anymore-screenly-resin-io?page=1#community_comment_360001891733)

rusko124, 2018-09-10:
@vpetersson 
I looked at the comment. I confirm that this problem exists. It is necessary to up the gpu_mem to 64MB.

vpetersson, 2018-09-10:
@rusko124 Is there any way to programmatically tell Resin (in the build description perhaps) to use this config? Also, perhaps more importantly, were you able to make it work (even after reboot)?

vpetersson, 2018-09-10:
Got it!

rusko124, 2018-10-05:
@vpetersson 
At the moment I have not found a solution how to programmatically increase the value of gpu_mem. However, an increase to 64 gpu_mem is written at https://github.com/Screenly/screenly-ose/blob/master/docs/resin.md

davidgruhin, 2018-12-05:
so I deployed on Balena and followed the above fixes and I still have this issue:

Server Error: RAN: '/usr/bin/omxplayer /data/screenly_assets/c74f1afa3e9244fbbcc7c87fe12cf6b3 --info' STDOUT: /usr/bin/omxplayer.bin: error while loading shared libraries: libWFC.so: cannot open shared object file: No such file or directory STDERR:

vpetersson, 2019-02-21:
@rusko124 can you post on the Balena forum to see if it is possible?

JeffResc, 2019-02-22:
Can confirm this issue still persists. I followed the steps in [the above comment](https://github.com/Screenly/screenly-ose/issues/868#issuecomment-418631717). Should a pull request with these changes be submitted?

vpetersson, 2019-02-22:
Thanks @JeffResc. I've adde this to the backlog for the near future when @rusko124 will return to OSE.

rusko124, 2019-04-24:
@vpetersson Can we move this issue to Sprint 10? We have issue like #843, this issue will fit perfectly into this sprint.

vpetersson, 2019-04-24:
Done.

closed 2019-08-23 by stale[bot]