Screenly/Anthias #870: Screenly starts into Terminal

lukashaider, 2018-07-25:
#### Overview of the Issue

Every time I make the IP Address static I have a 80% chance that screenly ose starts into a terminal.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Download "2018-05-04-Screenly_OSE_4GB.zip" Release
* Flash img onto SD Card
* Boot from SD Card
* change to static IP Address "/etc/network/interfaces"
* reboot
* 3 Dots appear. ~80% chance that a terminal shows up after that (weirdly with a grey background)
* If i press STRG+ALT+F2 screenly starts normally (but I can't do this every time the pi boots) 

### Environment

* Raspberry Pi Hardware Version: Model B+, B and Pi Zero W (strangely not with the Pi 3)
* Screenly OSE Version: v0.15 (My Michelle). (I never managed to get Screenly working with "bash <(curl -sL https://www.screenly.io/install-ose.sh)", although the script saying screenly installed successfully.

vpetersson, 2018-07-25:
I've seen something similar with older Pis. We'll see if we can take a look at that in the near future.

rusko124, 2018-10-01:
I reproduced this issue and made fixes.
- https://github.com/Screenly/screenly-ose/commit/72c96c3a70777639fb733f6a183cc761a7ce1a43
- https://github.com/rusko124/screenly-ose/commit/5fdb7624061c821336fa72cf59358a1f2491fba6
After these changes, I haven't a console at startup anymore.
@lukashaider Can you try to apply these edits and let me know if it help you?

closed 2018-10-22 by vpetersson