Screenly/Anthias #871: Stuck on Screenly Splash Screen with experimental update

bhagemann, 2018-07-25:
#### Overview of the Issue

Install a new image of Sprint7 on SD card.  Boot up on RaspPi.  Configure wi-fi and enable SSH.  Reboot the Pi via SSH.  Run the bash line to upgrade the card to the latest everything (experimental version, upgrade system and so on).  Reboot.  Blue Screenly screen comes up and stays.  Can't log in with web front end.  CAN connect with SSH, reboot it... does the same thing.
ait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version:  Tried on 3 and new 3+, exactly same thing happens
Tried with 2 SD cards, different brands
Tried with 2 power supplies

This only started happening about a week ago or so when had the experimental version running fine mostly, and ran the bash script to see if there were any dev updates it would pick up.  Since then I can't get an experimental version running.

vpetersson, 2018-07-27:
@antonmolodykh can you please take a look?

netjordanlee, 2018-07-30:
+1

Also getting this issue.

mscreations, 2018-07-31:
I've run into this issue as well. I was trying to see if the experimental version would work on my Pi Zero, and I found that the device boots and I can ssh into it, but none of the docker containers start up automatically. Issuing a `docker start screenly-ose-websocket && docker start screenly-ose-server && docker start screenly-ose-viewer` does take care of it though.

vpetersson, 2018-08-02:
@mscreations Interesting find. We have a new update for the Pi Zero in the pipeline, but I don't have an ETA for this yet. The other issue is a bit stranger and needs to be investigated.

antonmolodykh, 2018-08-02:
I confirm the problem. I don't understand why this is happening, because I did not change anything with regard to launching containers. I'll let you know when I fix it

mscreations, 2018-08-02:
The weird part is after i manually start the containers the first time, it works fine afterwards. I have had no issues with it starting afterwards. I do however have an issue where the whole pi locks up. No ssh, no web server response, and the viewer stops.

acruzmx, 2018-08-21:
+1 .. Same here, except that when I do docker start screenly-ose-websocket the container does not exist... Any idea?

vpetersson, 2018-08-22:
@acruzmx Did you try it with the latest changes to the experimental branch?

acruzmx, 2018-08-22:
Hi, thanks for asking. I was using the latest update. I ended up reinstalling the old one. 
The root problem I was trying to solve is that a couple of klipfolio dashboards are stalling an loosing connection, even though I can get to the console via SSL. I'm then able to reboot, over and over
Thanks again

rusko124, 2018-10-25:
Hi @acruzmx ! Do you still have this problem? If so, could you give the reproduce steps for this problem?

closed 2018-11-19 by vpetersson