Screenly/Anthias #874: crontab no available when using resin.io

dimiclic, 2018-07-27:
#### Overview of the Issue

Having Screenly OSE running on multiple devices with resin.io
Problem: crontab -e
    service is not installed and can not be installed both in host as in main

* Raspberry Pi Hardware Version: 3B
* Screenly OSE Version: 379c417

vpetersson, 2018-07-27:
This is expected as there is no crontab running inside the container. @antonmolodykh I think we should probably start moving to Celery or something similar to be able to process background tasks better.

dimiclic, 2018-07-27:
Are there plans to get crontab working in the near future?

vpetersson, 2018-07-27:
No, using a crontab in a dockerized environment is not the proper solution. This logic needs to be written with a worker (like Celery).

vpetersson, 2018-08-13:
Will be fixed in #876.

closed 2018-08-13 by vpetersson