Screenly/Anthias #876: Deprecate cronjobs in favor of worker

vpetersson, 2018-08-13:
As pointed out in #874, some features will not work in resin/dockerized environment. To solve this, we need to introduce a worker that can run inside a docker container.

I suggest that we use [Celery](http://www.celeryproject.org/) and port all cronjobs to run there..

Marco-LB, 2018-10-16:
Hello, is it possible to have a tutorial? instructions how to create a command to execute on startup? hdmi-cec on startup?

vpetersson, 2018-10-16:
@Marco-LB sure, take a look [here](https://www.screenly.io/blog/2017/07/02/how-to-automatically-turn-off-and-on-your-monitor-from-your-raspberry-pi/).

Marco-LB, 2018-10-16:
thanks for the quick reply, but cron and crontab know how to use them, but celery is the problem

vpetersson, 2018-10-16:
Well, this isn't implemented yet. That's what this ticket is tracking.

Marco-LB, 2018-10-16:
can this post help? [https://github.com/resin-io-playground/cron-example](url)

vpetersson, 2018-10-16:
We know how to do it. It's fairly straight forward. We just haven't implemented it yet. It is scheduled for the next sprint.

Marco-LB, 2018-10-16:
OK, Thanks for your information

rusko124, 2019-04-19:
This issue was solved in [Celery](https://github.com/Screenly/screenly-ose/tree/celery) branch(PR https://github.com/Screenly/screenly-ose/pull/987#issuecomment-450341787).
The subsequent discussion, I recommend to move here #1081

@vpetersson I think we can close this.

closed 2019-04-21 by vpetersson