Screenly/Anthias #879: Moved new installation to new network, powered on but no resin page to connect to new network

curbarthedog, 2018-08-16:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

I have just built a new Screenly installation on a Pi 3 B+. I set it up on one wifi network then moved it to a new location with a different wifi network. I am not getting the Resin WiFi hotspot page to connect it to the new network. The screen goes straight to the player and starts to display the loaded assets.

#### Reproduction Steps
Set up new device with latest image as per guidance, connect to one network, load assets, shut it down and move to different WiFi network

### Environment

Pi 3 B+

vpetersson, 2018-08-20:
@curbarthedog Did you you issue a "Reset wifi config" in the settings? If not, it is expected that the hotspot page will not appear.

curbarthedog, 2018-08-21:
I do now have this working by doing what you said.

Do you think this is the right behaviour? You could forsee the situation where an IT department changes the WiFi installation without notifying those with the Screenly setups and they would then be stuck?

Thanks for the help

vpetersson, 2018-08-22:
> Do you think this is the right behaviour? You could forsee the situation where an IT department changes the WiFi installation without notifying those with the Screenly setups and they would then be stuck?

Yes, this is the expected behavior. In short, there's no way (easy/reliable) way to differentiate between say a network outage (someone pulls the power of an access point) or a network change. Since we want to air on the side of caution and optimize for viewing experience, we do not want to throw up the dialog on the screen again (which also is a large security issue). 

We also recommend to always use a wired connection whenever possible.

Hope this makes sense.

closed 2018-08-22 by vpetersson