Screenly/Anthias #880: Unable to display rtsp feeds or streaming urls

hermy65, 2018-08-18:
Just installed ScreenlyOSE on a pi3 with a sunfounder 5" LCD to display a loop of a couple cameras using RTSP feeds. During install i tried normal, developmental and experimental and all 3 give me the same issue. 

I can pull the RTSP feed in on VLC without issue but when i add it to screenly only 1 will display the other two just cycle back to the screen that tells me what ip to go to so i can manage what is displayed.  I setup motioneye/motion and added my cameras there using the RTTSP feed and then tried using the streaming URL from motioneye which is just an http://ip:port which displays fine in my browser but in screenly it throws up a screen full of text as seen in the screenshot below. 

https://imgur.com/S6XTcgh

vpetersson, 2018-08-19:
@hermy65 For RTSP feeds, you must use `rtsp://` instead of `http://` for the player to be able to pick up that it is indeed an RTSP feed.

closed 2018-08-19 by vpetersson

hermy65, 2018-08-19:
I understand that, i said i tried both the RTSP feed AND motioneye's streaming url with is HTTP. Neither work and the streaming url from motion is just continually updating jpg's throws the screen in the image i attached. When using RTSP it just tells me what ip to use to modify the configuration

vpetersson, 2018-08-20:
@hermy65 RTSP is a bit of a mess unfortunately, as it covers a wide range of codecs etc. In short, the feed must be an h264 feed for it to work. The best way to troubleshoot this is to install a regular Raspbian version and then try to get the feed working using `omxplayer` (which is what Screenly OSE uses behind the scenes). 

I'd also recommend that you take a look at [ispyconnect](https://www.ispyconnect.com/sources.aspx), which includes a large database of various parameters etc for cameras. I've had it working with Axis cameras in the past, but even there you need to construct the correct URL for it to work.

hermy65, 2018-08-20:
@vpetersson When i pull my camera in through VLC it says t is using h264 as can be seen in the screenshot below. Is there any logging somewhere i can take a look at to see if there are any errors?

https://imgur.com/OzFdsiU

vpetersson, 2018-08-20:
No, unfortunately not. You'd have to debug it with `omxplayer`.