Screenly/Anthias #905: Unable to modify dates for assets in Firefox & Safari

luismtz9, 2018-09-17:
#### Overview of the Issue
We just updated to the newest master image and while the interface looks great it does not seem to work well with Safari and Firefox. In Safari, the Edit asset window loads but the date and time cannot be modified as the popup windows appear behind the asset window. Manually inputting a date does not work either can causes the web interface to freeze. In Firefox, I am unable to get the Edit Asset window to appear. It seems that clicking the pencil icon does not trigger that action. 

### Environment

* Raspberry Pi Hardware Version: Model 3 B+ 
* Screenly OSE: Master Image 
* Firefox: 62.0
* Safari: 11.1.2

vpetersson, 2018-09-18:
@luismtz9 Could you try upgrading to the latest development version and see if that resolves the issue? W'eve changed a fair amount of things since the last release as we are overdue for a new stable release.

luismtz9, 2018-09-20:
I updated to the latest image from the master branch (I assume this is the development version) today and it seems that I am unable to modify the dates in any browser now. The calendar & time popup's appear as expected however the dates are not saved once you click outside the popup. The start and end date then revert to a one month window from the minute I tried to modify it. This behavior is the same on Firefox, Chrome, & Opera. On Safari, the popups are still appearing behind the Edit Asset Window.

vpetersson, 2018-09-20:
ping @rusko124

vpetersson, 2018-09-21:
@luismtz9 Could you please upgrade to the latest development version? @rusko124 pushed a potential fix.

closed 2018-09-21 by vpetersson