Screenly/Anthias #909: Web assets get cached

abrenoch, 2018-09-19:
I am working on a page to report some internal metrics at our company, and in the process discovered that screenly appears to be caching the pages (or something to the same effect). I was not able to see the changes I had made until I restarted the pi.

To reproduce I simply added a locally hosted site, made some changes after screenly displayed the site (I did both js and html), and wait for the web page asset to be displayed again. I discovered the old version of the page I am working on was being displayed still. I rebooted and my changes were then available.

Running a pi 3 b+ with the "My Michelle" release.

Maybe an option to disable caching for web assets might be a good idea - If that is the intended behavior.

vpetersson, 2018-09-19:
@abrenoch Can you reproduce this if you switch to the experimental branch?

abrenoch, 2018-09-19:
@vpetersson I'll try to do that if I get the opportunity... Is there an easy way to switch back if there are any problems in that branch?

vpetersson, 2018-09-19:
Yes, you can just re-run the installer and switch back to the stable branch.

abrenoch, 2018-09-19:
Ok cool, I'll try to do that in the coming week!