Screenly/Anthias #918: ReactJs doesn't work

damienpeerbolte, 2018-10-01:
I'm trying to display a webpage with some reactjs included, but the reactjs part of the page doesn't work.

damienpeerbolte, 2018-10-01:
I already checked "y" for experimental branch at the installation of screenly.

damienpeerbolte, 2018-10-01:
It does behave the same way on both the branches

vpetersson, 2018-10-01:
What Raspberry Pi hardware?

Also, we would need the steps to reproduce this in order to look into it.

damienpeerbolte, 2018-10-01:
It's a raspberry pi 3b

damienpeerbolte, 2018-10-01:
But if I install Raspbian Stretch(with Desktop) on the Pi and then go into Chromium, the webpage I want to display works. If install screenly directly on the pi(without raspbian) the reactjs part of the webpage won't work.

vpetersson, 2018-10-01:
Yeah, Chrome does things a bit different. However, if we are able to reprocude, we can look into it.

damienpeerbolte, 2018-10-01:
that means it's the browser that doesn't accept the reactjs, not the pi, right?

vpetersson, 2018-10-01:
Not sure. We haven't tested it very closely. Could maybe be some setting.

damienpeerbolte, 2018-10-01:
Also when I install screenly over raspbian it works, how come it doesn't work on screenly only?

damienpeerbolte, 2018-10-01:
oh wait, it's installed on lite without a desktop.. makes sense

closed 2018-11-26 by vpetersson