Screenly/Anthias #921: Can't get into terminal after screenly manual installation(on raspbian)

damienpeerbolte, 2018-10-02:
I have installed screenly ose manually with the command in raspbian. But when I boot screenly, I can't get into the terminal(CTRL+ALT+F1)

vpetersson, 2018-10-02:
Try TTY2 (ctrl+alt+f2).

damienpeerbolte, 2018-10-02:
That also doesn't work, and I can't click with the right mouse button to open for example 'Inspect Element'

damienpeerbolte, 2018-10-02:
I can open the terminal when it's still booting but a few seconds later when screenly boots fully, screenly will get on top of everything

vpetersson, 2018-10-02:
Ah, so you're using the experimental branch. Yes, that is expected since Screenly takes over the frame buffer. You need to use SSH.

damienpeerbolte, 2018-10-02:
That works!! But unfortunately I still got the ReactJs problem..

vpetersson, 2018-10-02:
If you provide complete steps to reproduce the issue, we might be able to look at it next sprint.

damienpeerbolte, 2018-10-02:
I've done the following:

- Installed Raspbian(with desktop) on sd card using etcher
- followed all the steps on the screenly ose website(the manual installation with the command) from top to bottom
- put in the website link in the web interface

On raspbian when screenly wasn't installed yet, the reactjs part of the webpage I want to display in screenly did work! But when I put in the exact same url into screenly, it doesn't work.

vpetersson, 2018-10-02:
That doesn't help unfortunately. We need the exact website (or a similar website that behaves the same way) in order to test it.

damienpeerbolte, 2018-10-02:
I'm trying to display [this page](https://panels.athlinks.com/ct/panel/5b9a1f911849f). Screenly shows the page but it won't load the reactjs part. Also for example, screenly won't display [this demo page](http://ngokevin.github.io/aframe-react-boilerplate/).

vpetersson, 2018-10-02:
Changing title since it is expected that you can't access the terminal with the experimental branch.

vpetersson, 2018-10-02:
I would be *very* surprised if you would be able to get [this page](http://ngokevin.github.io/aframe-react-boilerplate/) to work on the Raspberry Pi. It is *far* too heavy on animations etc.

damienpeerbolte, 2018-10-02:
Yeah, I kinda knew that already(it was a bad example). But it should display the one I want to display, right?(a co-worker of mine told me that it worked once)

vpetersson, 2018-10-02:
Maybe, but just that by itself is pushing it. We've done a lot more work in Screenly Pro to accelerate playback. If you want to hack on the web playback performance, the source code browser is available [here](https://github.com/Screenly/screenly-ose-webview).

damienpeerbolte, 2018-10-05:
Screenly is currently using UZBL as browser, right? How can I change that to for example Epiphany or maybe Chromium?

vpetersson, 2018-10-05:
> Screenly is currently using UZBL as browser, right? How can I change that to for example Epiphany or maybe Chromium?

The 'stable' and 'development' branch are both using UZBL as a browser. That is correct. 

Switching to Chromium or Epiphany is unfortunately not an option, since neither provide a framework for remote control. There's a long conversation about that in #527. The tl;dr is that the experimental branch is now running a different browser.

damienpeerbolte, 2018-10-05:
Ok, I'll upgrade screenly to experimental(currently development branch) to see if it will work,

damienpeerbolte, 2018-10-05:
I've upgraded screenly to the experimental branch and the reactjs part of the page still doesn't work.

vpetersson, 2018-11-26:
Ok, we need a copy of the React page in question to see if we can reproduce it. Without that we can't do anything.

closed 2018-11-26 by vpetersson

bglidwell, 2021-06-13:
Also unable to load React pages. In my case it won't even render a simple login page with image background. No fancy animations or anything, just a CMS portal. I see this issue is closed, is there any hope for this being available on future versions or should I look into other solutions?

vpetersson, 2021-06-15:
@bglidwell Are you using the developer version or stable version?

bglidwell, 2021-06-15:
> @bglidwell Are you using the developer version or stable version?

Stable

vpetersson, 2021-06-15:
Switch to the developer version and see if that works as it's using a different browser.

bglidwell, 2021-06-25:
I installed RaspberryPi OS Lite and ran the installer from the bash command you have listed. Chose 2 to install Developer version. It boots, I see the 3 dots screen, then it dumps to a login prompt...