Screenly/Anthias #929: Cannot change WiFi networks even with Development upgrade.

ssantora002, 2018-10-08:
Overview:
I'm trying to connect to a different SSID. The 'Reset Wifi Settings' that is on the Screenly Settings page is not working properly. I moved the Pi to a new office where the WiFi is different. The  

Reproduction Steps:
- Downloaded and verified hash of 2018-05-04-Screenly_OSE_4GB.zip
- Fresh install of Screenly-OSE from github image 2018-05-04-Screenly_OSE_4GB.zip to SD Card
- Connected to devices hotspot and configured wifi
- Upgrade to development version
- Powered down pi after successfully upgrading.
- Move pi to another office with different SSID
- The device boots and when it gets to the splash screen where the hotspot info is supposed to display I only have a blank white screen.

- I moved it back to the original wifi location and it connected fine and displayed an IP
- I went to the screenly web page and selected settings. I selected Reset Wifi Settings. It said the request was sent successfully.
- I powered down the Pi and moved it to the new Wifi network again. 
- The device boots and when it gets to the splash screen where the hotspot info is supposed to display I only have a blank white screen.

Environment
Raspberry Pi Hardware Version: Raspberry Pi Zero w/WiFi on 16GB micro sd card
Screenly OSE Version: latest github image with latest screenly ose upgrade: 2018-05-04-Screenly_OSE_4GB.zip

Please help. Thank you.

zasnicoff, 2018-10-24:
I don't know if screenly will have problems with you editing these files, but I did it to get my IP address static and it worked. 
```
/etc/network/interfaces
/etc/wpa_supplicant/wpa_supplicant.conf
```

You can add:

```
network={
    ssid="ESSID"
    psk="Your_wifi_password"
}
```
to the latter and it might solve your WiFi issue.

For more information, also check:
https://raspberrypi.stackexchange.com/questions/37920/how-do-i-set-up-networking-wifi-static-ip-address

rusko124, 2018-10-26:
Hi, @ssantora002 !
Can you try update one more time?
A recent changes has been fixed with a blank white screen problem.

ssantora002, 2018-10-28:
Rusko,

I appreciate your effort to help me solve this issue. I haven't attempted
anything at this time. I write you to let you know what my steps will be to
try this fix.

   1. Dowload latest github release. 2018-05-04-Screenly_OSE_4GB.zip and
   verify hash.
   2. format SD card and install fresh image of latest github release of
   screenly ose, 2018-05-04-Screenly_OSE_4GB.zip.
   3. Boot up screenly on raspbery pi zero w/wifi and connect to SSID with
   Resin Wifi Connect (note: I've tried connecting Ethernet wire vis USB and
   doing the upgrade in the past but when I remove the USB and reboot the
   device the white screen happens too),
   4. Once I'm connected to the internet I run this command,
   sudo ~/screenly/bin/run_upgrade.sh
   5. I select No to everything except I answer Yes to install development
   mode.
   6. Reboot.

Most of the time I seem to keep having the issue where if I reboot, even
after installing the development mode, it will say 'Unable to resolve local
IP address'. I suspect that this is because I have to connect to the
internet to do the upgrade and after the upgrade the files were already
created so this prompts the wifi connect not to launch. Another Important
note is that I definitely cannot edit the wpa_supplicant file or use a
network.ini file in the root folder. Neither of these tricks have worked
since the 2018-05-04-Screenly_OSE_4GB.zip version was released. I do have
older versions of screenly and they still work fine. I'm thinking about
just keeping these versions running because I am beginning to get
frustrated with this 2018-05-04-Screenly_OSE_4GB.zip version. Nothing is
working as intended with it in regards to the WiFi, the 'Reset WiFi
Connections" fails to work for example. I've tried NOOBS too and no
difference because it is just downloading the same image from the githum
repository. I've tried almost everything I can think of and I still cannot
solve the issues/ Also removing the wifi_set file doesn't work because I
cannot find it this file. I've checked if it is hidden too and nothing.
I've also tried to run the start_resin_wifi.py script on it's own and it
returns errors.

This was your proposed solution:
Can you try update one more time?
*A recent changes has been fixed with a blank white screen problem.*

Please let me know when these recent changes took place. Again, I haven't
tried again since late September of this year. I will try the upgrade again
and report back to you what happens.

rusko124, 2018-10-29:
@ssantora002 
I reproduced all the steps that you wrote. And in my case, everything works well.

ssantora002, 2018-10-29:
By everything did you try the wifi reset feature and then attempt to
connect to a different SSID?

rusko124, 2018-10-29:
@ssantora002 try reproduced this steps:
- Delete file initialized from /home/pi/.screenly/ or click reset wifi in web interface
- Remove or replace files from /etc/NetworkManager/system-connections/
---------
Since you did not choose manage network when upgrade(You can skip next steps if agreed.) :
- sudo cp /home/pi/screenly/ansible/roles/network/files/wifi-connect.service /etc/systemd/system/
- sudo systemctl enable wifi-connect
- rm /home/pi/.screenly/disable_manage_network
--------
- Reboot

ssantora002, 2018-10-30:
Should I be installing Network Manager? Please described the what this
option does? Please describe the benefits or possible disadvantages of
installing this option. I could not much document ation on it. Any info you
can provide is much appreciated.

vpetersson, 2018-10-30:
tl;dr - Network Manager is the modern way to deal with networks on Linux and is reqiured for the hotspot feature. For pros/cons, please consult Google :)

ssantora002, 2018-10-31:
I installed the latest Github release of screenly-ose and I ran sudo
~/screenly/bin/run_upgrade.sh
I installed with Y to development and Y to network manager.
After the reboot I still get the 'unable to connect to local ip' message.

After installing upgrade I tried these steps that were provided by your
support team below and it solved the issue.

   - Delete file initialized from /home/pi/.screenly/ or click reset wifi
   in web interface
   - Remove or replace files from /etc/NetworkManager/system-connections/
   - sudo cp
   /home/pi/screenly/ansible/roles/network/files/wifi-connect.service
   /etc/systemd/system/
   - sudo systemctl enable wifi-connect
   - rm /home/pi/.screenly/disable_manage_network

 Important note: the last step said the file could not be found. I check my
spelling and ran an ls command to check the folder. It was not there.
Regardless, the issue is solved. I can now set my configurations and create
a custom image.

Thank you 😀

vpetersson, 2018-11-26:
Closing out the issue as it appears to have been resolved.

closed 2018-11-26 by vpetersson