Screenly/Anthias #933: There is no possibility to add a page with requiring js

rusko124, 2018-10-10:
#### Overview of the Issue

The problem is described in detail https://support.screenly.io/hc/en-us/community/posts/360017643434-URL-error-for-asset.
I reproduced this problem and can confirm it.
[image]
When screenly check on validate this url, we send GET request and receive the 404 status code, this is due to the fact that javascript is disabled.
@vpetersson Do we need to solve such situations? If yes, then I can suggest using something like this https://github.com/kennethreitz/requests-html

vpetersson, 2018-10-10:
I think that overcomplicates things. Basically, that's a bug in the webiste/server rather than in Screenly. It should return a 20X. Perhaps a better option is to add an option for something like "Skip asset check."

rusko124, 2018-10-10:
Got it!

vpetersson, 2018-10-11:
Closing out. Please ping the original reporter to ensure it worked.

closed 2018-10-11 by vpetersson