Screenly/Anthias #941: Improvement idea

genvagula, 2018-10-21:
#### Screen area change

Would like to set certain area from the total display for the screenly to display the content.
For example i have big LED display that has area of 384x192 px.  So i need to show content at this area.
Cannot find at the moment that capability.

vpetersson, 2018-10-22:
@genvagula That's unforutnately not a [supported resolution](https://elinux.org/RPiconfig) for the Raspberry Pi, so that's unfortunately not something we would be able to supprt.

closed 2018-10-22 by vpetersson

genvagula, 2018-10-22:
thnx for answer. but i thought that, for example video output would be 1280x768 but the output play area would be set at the upper left corner 384x192px and i could resize that area at the user interface.

vpetersson, 2018-10-22:
That's not something that is easily attainable. It would require a fairly significant rewrite.