Screenly/Anthias #943: rsycn one player to another

bobsol, 2018-10-22:
I am trying to rsync one player with another.  Player doesn't play the changed assets and db unless I `systemctl restart screenly-viewer`.

Details:

I have am rsyncing both `screenly_assets` and `.screenly/screenly.db` 

As an alternative, i tried rsyncing the whole `screenly` folder. No improvement.

`update_playlist` and `refresh_playlist` seem to be running.

I have checked with sqlight3 that the database is actually copying over.

Restarting `screenly-viewer` is too disruptive to the display to do as often as I would like.

Any help?

thanks in advance.

### Environment

* Raspberry Pi Hardware Version:  3B+
* Screenly OSE Version: 2018-05-04-Screenly_OSE_4GB.zip

vpetersson, 2018-10-22:
That's not really a supported use case. If you want to do something like that, you should write something that uses the API.

closed 2018-10-22 by vpetersson

bobsol, 2018-10-24:
Here is a diff if you care.

--- viewer.py.orig	2018-10-24 14:34:22.076759685 -0400
+++ viewer.py	2018-10-24 14:33:24.427157304 -0400
@@ -155,6 +155,12 @@
         time_cur = datetime.utcnow()
         logging.debug('refresh: counter: (%s) deadline (%s) timecur (%s)', self.counter, self.deadline, time_cur)
         if self.get_db_mtime() > self.last_update_db_mtime:
+	    # testing: reload database
+	    global db_conn
+	    if db_conn != None:
+		db_conn.close()
+	    db_conn = db.conn(settings['database'])
+	    #
             logging.debug('updating playlist due to database modification')
             self.update_playlist()
         elif settings['shuffle_playlist'] and self.counter >= 5:
@@ -164,12 +170,15 @@
 
     def update_playlist(self):
         logging.debug('update_playlist')
-        self.last_update_db_mtime = self.get_db_mtime()
+	# mtime save was here
         (new_assets, new_deadline) = generate_asset_list()
-        if new_assets == self.assets and new_deadline == self.deadline:
+	# added check to db_mtime
+        if new_assets == self.assets and new_deadline == self.deadline and self.last_update_db_mtime == self.get_db_mtime():
             # If nothing changed, don't disturb the current play-through.
+	    self.last_update_db_mtime = self.get_db_mtime()
             return
 
+        self.last_update_db_mtime = self.get_db_mtime()
         self.assets, self.deadline = new_assets, new_deadline
         self.counter = 0
         # Try to keep the same position in the play list. E.g. if a new asset is added to the end of the list, we

vpetersson, 2018-10-30:
@rusko124 Can you look at the diff from @bobsol? My comncern is largely about added I/O operations, which will in turn reduce the life span of the devices.