Screenly/Anthias #949: Installation is failing within the ansible playbook

offlinehoster, 2018-10-26:
> TASK [screenly : Create default assets database if does not exists] ************************************
> An exception occurred during task execution. To see the full traceback, use -vvv. The error was: ImportError: cannot import name to_bytes
> fatal: [localhost]: FAILED! => {"msg": "Unexpected failure during module execution.", "stdout": ""}
> 	to retry, use: --limit @/home/pi/screenly/ansible/site.retry
> 
> PLAY RECAP *********************************************************************************************
> localhost                  : ok=46   changed=12   unreachable=0    failed=1   
>

closed 2018-10-30 by vpetersson

offlinehoster, 2018-10-30:
Problem is still existing. Just for your information. Installed on a current raspberian lite image.

vpetersson, 2018-10-30:
Please provide these details, as we won't be albe to provide assistance otherwise:

```
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

A paragraph or two about the issue you're experiencing.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version:
* Screenly OSE Version:
```