Screenly/Anthias #951: installation on raspian failing task screenly : Create default assets database if does not exists

pooh22, 2018-10-30:
#### Overview of the Issue

I was trying the manual installation method with the shell script and ansible. The problem happens at the end (I presume) of the install. Output is:

```
TASK [screenly : Create default assets database if does not exists] *********************************************************************************************************************************************
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: ImportError: cannot import name to_bytes
fatal: [localhost]: FAILED! => {"msg": "Unexpected failure during module execution.", "stdout": ""}
        to retry, use: --limit @/home/pi/screenly/ansible/site.retry

PLAY RECAP ******************************************************************************************************************************************************************************************************
localhost                  : ok=52   changed=34   unreachable=0    failed=1   

```

#### Reproduction Steps

* take clean raspian (v9)
* turn on ssh
* raspi-config (change user password, extend filesystem)
* reboot
* install vim
* edit /etc/locale.gen (add nl_NL.utf8) and run locale-gen
* apt update && apt upgrade
* run  (copied from history)
```
    2  curl -sL https://www.screenly.io/install-ose.sh >install-ose.sh
    3  bash install-ose.sh
    4  sudo apt install -y network-manager
    5  bash install-ose.sh
```
### Environment

* Raspberry Pi Hardware Version: 2
* Screenly OSE Version: git master commit 62bfe2ff477122ee97bf1b1017e94abffad52013

rusko124, 2018-10-31:
Hi, @pooh22 !
Please tell me the version ansible installed, and what version ansible installs the install-ose.sh.( String **sudo pip install ansible==VERSION**)

pooh22, 2018-10-31:
Hi @rusko124, I seem to remember ansible 2.6.2 or 2.6.8 and the version of screenly/ose was stable master, not experimental or development.
I didn't have a spare rpi available, so I used one that is normally in use for something else... It will be a while before I have another opportunity to test (it takes a really long time to install on my rpiv2)

BTW, I tried installing twice; first time I tried without network managed by screenly, got the error, so I tried again the next day, on a clean raspian and got the same error. That's why I decided to post the issue here, since it seems to be a real bug in the installer (at least, on the current raspian on my rpi)

Cheers
/Simon

rusko124, 2018-10-31:
@pooh22 
I think I know what the problem is.
The best solution to this problem is to install the stable version from here https://github.com/Screenly/screenly-ose/releases/tag/v0.15 or using NOOBS.
But if you still want to use install-ose.sh, you need to change in the script:
**sudo pip install ansible==2.6.3**
on it
**sudo pip install ansible==2.1.0.0**

rusko124, 2018-10-31:
At the moment install-ose.sh may have some problems with the installation of production branch. We know this, but as Sprint 8 comes to an end, this problem will disappear with the release of a new stable version of Screenly-OSE.

pooh22, 2018-10-31:
OK, I'll take your word for it then ;-) I'm looking forward to trying the next release...

memorex258, 2018-11-20:
> @pooh22
> I think I know what the problem is.
> The best solution to this problem is to install the stable version from here https://github.com/Screenly/screenly-ose/releases/tag/v0.15 or using NOOBS. [...]

Thank you, this allowed me to get passed that part when using the install-ose.sh script

vpetersson, 2018-11-26:
@pooh22 just as a heads-up - the new release is live now, so let us know if it is still an issue. I'll close out the issue and re-open it if it is still an issue.

closed 2018-11-26 by vpetersson