Screenly/Anthias #953: Viewer crashed with experimental branch

Amosar, 2018-11-07:
#### Overview of the Issue

I have tried screenly-ose with uzbl but it doesn't support video so I have wanted to try the experimental web browser but it crash when I launch screenly.
I have this error: https://pastebin.com/DranEY9w

#### Reproduction Steps

* Flash the raspberry with screenly image
* switch to the experimental branch and run the update script
* restart the raspberry
* And the viewer will crash at start

### Environment

* Raspberry Pi Hardware Version: raspberry Pi 3 Model B
* Screenly OSE Version: git, branch experimental, commit b34de4d748158a37c628103487a624edce49f49b

rusko124, 2018-11-07:
Hello @Amosar!
Can you tell me if you had any errors with the update, please?
You don't need to switch to the experimental branch, just run the update script and select experimental when asked. I think that might be the cause of your problem.

Amosar, 2018-11-07:
Yes, the problem have been cause in the installation by bad proxy settings, sorry for opening this issue

closed 2018-11-07 by Amosar

rusko124, 2018-11-07:
I'm glad your problem has solved. Have a good day!