Screenly/Anthias #956: Bad encoding in setting player name

teddiegp, 2018-11-14:
#### Overview of the Issue

In the web UI, I set "Player name" to something containing the german letter ß. When saving, a error 500 page was shown. Afterwards, all settings were reset (including [auth] user, password).
I tried again several times and the problem only occurred when I used german umlauts (ä ö ü ß).

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Open the Screenly Web Interface
* Click "Settings"
* In the field "Player name", input something containing a umlaut
* Save

### Environment

* Raspberry Pi Hardware Version: 3B+
* Screenly OSE Version: 2018-05-04-Screenly_OSE_4GB.zip

vpetersson, 2018-11-23:
Should now be resolved in the development version.

closed 2018-11-23 by vpetersson

teddiegp, 2018-11-28:
Not sure whether to reopen the task, since the patch got reverted (#967).

vpetersson, 2018-11-28:
@teddiegp it got reverted because it broke other things. An improved version has been merged in (#970).

teddiegp, 2018-11-28:
I didn't see that one, thanks!