Screenly/Anthias #957: Raspberry Pi 3 B+

DudleySmith, 2018-11-14:
#### Overview of the Issue

I did not succeed to setup WiFi connection:

#### Reproduction Steps

* not permanent acces point
* if finally succeed to connect to acces point, and succeed to load the wifi setup page, the DropDown List is emtpy of SSIDs
* If locally (wire) connected and click on reset WiFi configuration , I got an "unable to restore local IP adress
* still I can manage my screen with local/wire

### Environment

* Raspberry Pi Hardware Version: 3 B+ brand new
* Screenly OSE Version: last release, v.0.15.1

rusko124, 2018-11-23:
Hi, @DudleySmith !
You can see possibly useful information for you here #929!
Let me know if it helped you or not, please.

vpetersson, 2018-11-23:
@DudleySmith - Could you test with the new image that we just released?

vpetersson, 2018-12-11:
Closing out this issue due to lack of response.

closed 2018-12-11 by vpetersson