Screenly/Anthias #964: Over heating with - SPrint 8 0.16 version

shibukraj, 2018-11-23:
Team

#### Overview of the Issue

Today updated the Screenly OSE to the latest version of 0.16. Ran for less than 30 min and it started showing the thermometer on the screen -- meaning overheating of the device. I was running the earlier version of the Screenly for days and I have never seen this. Just want to report this out.

#### Reproduction Steps

Install the latest version 0.16 from the Github.

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL 
I am using an Internal URL.But any URL will do
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version:
Rasberry Pi 3.0 

* Screenly OSE Version:
0.16 version

vpetersson, 2018-11-23:
@shibukraj you need to be more specific in the types of assets you are using.

vpetersson, 2018-11-23:
Also, see #965 if you used the disk image.

shibukraj, 2018-11-23:
Thanks . I will update to the latest build and will let the team know the results. Another think I noticed in the while splash screen after the assets move to another. My assets are mainly Grafana dashboards with no authentication and running under http.

vpetersson, 2018-11-23:
> Thanks.   Do we have any articles on how to switch to the experimental branch.  Wont mind testing that out and letting the team know about the  performance.

Simply run the the upgrade script as outlined [here](https://www.screenly.io/ose/) and answer yes to the question if you'd like to run the experimental version.

> We are in the process of going Pro version after we do a proof of concept with the OSE version.

Cool. Pro and OSE doesn't share any code base, so it's a very different system.

closed 2018-11-26 by vpetersson