Screenly/Anthias #965: Errno 13 permission denied

klarmacker, 2018-11-23:
[image]

Hello

Iam quiet new in the Screenly business. i set it up and the first thing i tried was to upload a Picture. 
But i stuck and get the Error in the Picture above.

Can someone help me?

Thanks
chris

vpetersson, 2018-11-23:
hi @klarmacker - interesting. Is this on the latest disk image?

vpetersson, 2018-11-23:
I just reproduced it - will issue a fix shortly.

vpetersson, 2018-11-23:
The temporary solution is to do the following:
 * Login using the default credentials
 * Run `chown -R pi:pi screenly_assets`

vpetersson, 2018-11-23:
Bug fixed and new disk image is being built (but it will take a few hours).

vpetersson, 2018-11-23:
This has now been fixed in the latest disk image. Thanks for reporting.

closed 2018-11-23 by vpetersson

klarmacker, 2018-11-24:
ok great thanks i will update it on Monday

klarmacker, 2018-11-26:
Working now thanks for your Service