Screenly/Anthias #969: Stops playing or crash after loading webpage

TheCazz, 2018-11-27:
Hi I have notice something strange with the latest and the version before that some webpage make screenly just stop working. It show the webpage all the time and I have to do a reboot,

Some page make it crash and instead of show the page it show screenly start screen with the ipaddress and then go to the imgge until is time to show the webpage again it crash and show the screenly start screen.

But some webpage works fine and I can't think of why??

I have also see when it have problem to show the default clock or weather page it stop working too.

I run with a Raspberry Pi B+ with the latest screenly OSE image that I have download from github

rusko124, 2018-11-27:
Hi, @TheCazz !
Can you provide the url of the web pages you use, please?

TheCazz, 2018-11-27:
Hi again and thanks for the very fast replay
I Think I understand the problem (Maybe) and that is I'm from sweden and we have å ä ö and in the error log I can see "ascii codec can't decode"
I can give you the whole screenly-viewer but I is going to be a very long one :)

I did try with this page to see what kind of meal a local school is going to have.
https://skolmaten.se/asenskolan/

The other pages that I have problem is on a local server so is not easy for me to show them to you

TheCazz, 2018-11-27:
[errorlog.txt](https://github.com/Screenly/screenly-ose/files/2619342/errorlog.txt)

Here is the copy and paste log

vpetersson, 2018-11-27:
@TheCazz looks like you're spot on. We should definitely support unicode characters. Assigning this to the coming sprint.

TheCazz, 2018-11-27:
Did try with the webpage that stop at the webpage and show nothing more then just the webpage but the log does not say so much 

Nov 27 10:36:46 raspberrypi python[597]: Sleeping for 10
Nov 27 10:36:56 raspberrypi python[597]: Generating asset-list...
Nov 27 10:36:57 raspberrypi python[597]: Showing asset http://addresstothewebpage.php?info=http://pagetotherssinfo/rss/days/ (webpage)

It maybe does not like that I have two http address to connect to the webpage??
The first address is to the PHP server on the local server and the second one is to a webpage that have a RSS feed

TheCazz, 2018-11-27:
Ahh nice thanks. 
Looking for the update :)

TheCazz, 2019-01-15:
Just curious about how is going. I mean I'm curious to try and like to know if it fix in the latest development version or how I can see what is new in the development version. New to this so sorry if it sound strange :)

ealmonte32, 2019-03-06:
@TheCazz -

Can you try to do what I suggested for the other user at this forum post and let me know if that for the time being works for you?

https://forums.screenly.io/t/crash-an-unexpected-tls-packet-was-received/36/5

brand1417, 2019-03-22:
I don't know if this is the correct place to chime in but I have the NOOBS installed screenly and I can play video just fine but most image or web assets seem to trigger the crash/restart.

vpetersson, 2019-03-22:
@brainrape do you have steps to reproduce that?

brand1417, 2019-03-22:
I *think* you meant to @ mention me... but I'll go with brainrape, too.

- clean install of screenly via noobs

brand1417, 2019-03-22:
I think you meant to mention me but I also will accept brainrape.

Using this image https://imgur.com/a/fIEcZjs#V7vVOBn and alternating it between that and any movie asset seems to do it. I also tried adding http://www.dexafit.com as a URL asset with similar results. Clean install of screenly via NOOBS. Tested on multiple Pi's ranging from 1-3 and Zero W.

ealmonte32, 2019-03-22:
Lol @ brainrape, yeah  he probably meant to do @brand1417 ..

So I tested both of the URLs you give and my screenly viewer/uzbl process both and load just fine..

I dont do NOOBS, I run Raspbian-Lite and install screenly via bash command, then do all the usual apt update and full-upgrade etc, and rpi firmware update and I have not issues with browser restarting or crashing..

can you provide the following:
```
raspbian/pi version, commands:
cat /sys/firmware/devicetree/base/model
cat /etc/debian_version

uzbl version:
dpkg -l | grep uzbl

libwebkit version:
dpkg -l | grep libwebkit
```

That would help, thanks.

brand1417, 2019-03-22:
I will get you the output of those commands later today and then attempt to do it via raspbian-lite as you suggest. It would be disheartening if the noobs installer wouldn't work as expected since, ya know, it's supposed to.

rusko124, 2019-04-19:
@vpetersson 
The issue has been resolved. 
Solution: upgrade to the development branch.
I think we can close the issue.

closed 2019-04-20 by vpetersson