Screenly/Anthias #971: Add support for displaying the Raspberry Pi version/model in the user interface

vpetersson, 2018-11-27:
I'd like to add the ability to display the Raspberry Pi version in the system info under System Info.

We can get the hardware version from here:

```
$ grep Revision /proc/cpuinfo
Revision        : a01041
```

Then we simply need to parse this, and turn it into the model number using [this list](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md).

dig1talmad, 2019-01-07:
While you're at it, could you display the version of Screenly in System Info too?

vpetersson, 2019-01-08:
Good idea @dig1talmad.

qbicdesign, 2019-04-17:
yay

qbicdesign, 2019-04-17:
can we get MAC address(es) added too please?

vpetersson, 2019-04-17:
@qbicdesign Yes shouldn't be very hard. I guess we need to display both for WiFi and Ethernet.

rusko124, 2019-04-26:
Raspberry Model / Screenly Version / MAC address were added to the system info page.
@vpetersson I think we can close this.

closed 2019-04-26 by vpetersson