Screenly/Anthias #974: When clock or weather stop working everything stop working

TheCazz, 2018-11-28:
Hi
I did write a issues yesterday about loading webpage make it crash
I did mention that it was something wrong sometime with the clock or weather sometime and now I have the log and a picture. 

```
Nov 28 09:59:13 raspberrypi python[594]: Asset Screenly Weather Widget at https://weather.srly.io is not available, skipping.
Nov 28 09:59:14 raspberrypi python[594]: Asset Screenly Clock Widget at https://clock.srly.io is not available, skipping.
Nov 28 09:59:14 raspberrypi python[594]: Asset Screenly Weather Widget at https://weather.srly.io is not available, skipping.
Nov 28 09:59:15 raspberrypi python[594]: Asset Screenly Clock Widget at https://clock.srly.io is not available, skipping.
Nov 28 09:59:16 raspberrypi python[594]: Asset Screenly Weather Widget at https://weather.srly.io is not available, skipping.
Nov 28 09:59:17 raspberrypi python[594]: Showing asset Screenly Clock Widget (webpage)
Nov 28 09:59:18 raspberrypi python[594]: current url is https://clock.srly.io
Nov 28 09:59:18 raspberrypi python[594]: Sleeping for 10
Nov 28 09:59:28 raspberrypi python[594]: Showing asset Screenly Weather Widget (webpage)
Nov 28 09:59:29 raspberrypi python[594]: current url is https://weather.srly.io
Nov 28 09:59:29 raspberrypi python[594]: Sleeping for 10
Nov 28 09:59:40 raspberrypi python[594]: Showing asset Screenly Clock Widget (webpage)
Nov 28 09:59:40 raspberrypi python[594]: current url is https://clock.srly.io
```

It look at the log that everything is fine after hte problem to show weather or the clock but the screen still show this 
[image]

and nothing more is happen and I have to reboot the raspberry pi.

Cryjack, 2018-11-28:
confirm. same here.

vpetersson, 2018-11-28:
I'm unable to reproduce this unfortunately.

TheCazz, 2018-11-28:
Hmm then that is a problem. Right now I just remove the weather and the clock but what if another webpage is on the screenly and they have problem with the webserver so the page goes down what happen to the screenly?

vpetersson, 2018-11-28:
Normally, that won't be a problem, as we check for a 20x HTTP status before we bring it up on the screen.

TheCazz, 2018-11-28:
hmm ok so make it sure it works is just remove the weather and the clock widget.

willworkforicecream, 2018-11-28:
Also happened to me a few times yesterday. Haven't seen it happen to other web assets, just the weather or clock widgets.

TheCazz, 2018-11-28:
Maybe when it show that "Unable to load page" is does not give any 20x HTTP status so the script can't see is does not working??

Cryjack, 2018-11-29:
Got the same.

[image]

Problem is a TLS error. The Page gives HTTP 200 Response, so there was no problem showing it. But this stops all other assets from beeing showed.

vpetersson, 2018-11-29:
Yeah so the problem here is two-fold: 

 * We need to investigate *why* this TLS error appears.
 * We need to make sure that even if it does appear, we need to gracefully skip it.

rusko124, 2018-11-30:
It should be decided now, at least when the viewer are stuck. Can I get a feedback, if it has helped?

TheCazz, 2018-11-30:
Well I can but not until Monday for me and then I going to try.

bhagemann, 2018-12-17:
Same happening here with new install of latest (11-23) image, also did the update over ssh (only the normal update, not to development or experimental).  Today (12-17) it gave this same error again.

Should I be doing development (or experimental) updates instead to get latest fixes?

Thanks!

vpetersson, 2018-12-17:
@bhagemann If you ran an upgrade from that version and didn't select the development version, you essentially "upgraded" to the same version you are running. Hence you need to switch to the development version to test the latest updates.

bhagemann, 2018-12-17:
thanks vpetersson... I'm running the dev branch update now.  will report back if anything notable.

ealmonte32, 2019-01-10:
@rusko124 
I was able to also reproduce and right away stopped viewer to look at logs and just wanted to show you, do you think TLS error has to do with sync or time happening at the same time as browser is loading the page?

Since you already changed the viewer.py file with adding LOAD_ERROR, should we bother to investigate this TLS handshake error or not waste time?

[image]

TheDutchEdo, 2019-01-14:
I got this with the clock widget

python[558]:  /usr/local/lib/python2.7/dist-packages/urllib3/connectionpool.py:847:  InsecureRequestWarning: Unverified HTTPS request is being made. Adding  certificate verification is strongly advised. See:  https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
--
python[558]:   InsecureRequestWarning)

rusko124, 2019-01-14:
Hi @TheDutchEdo !
Could you try upgrade Screenly to development version?

TheDutchEdo, 2019-01-14:
Still the same error, @rusko124

TheDutchEdo, 2019-01-15:
Never mind still got this:

Problem occurred while loading the URL https://clock.srly.io/
Error performing TLS handshake. An unexpected TLS packet was received.

And this in my log:

python[558]: /usr/local/lib/python2.7/dist-packages/urllib3/connectionpool.py:847: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings

python[558]: InsecureRequestWarning)

ealmonte32, 2019-01-16:
@rusko124 
Do you think this doesnt have to do with time on system during the connection? I think it might.
I am getting it now at every reboot (see screenshot), but after i restart the screenly.service it works because the time has been synchronized already by that time, maybe?

see screenshot of syslog before and after restarting just the screenly.service and not the rpi.

before, right after reboot:
[image]

after, just restarting screenly.service:
[image]

ealmonte32, 2019-02-07:
@TheCazz @TheDutchEdo @bhagemann @Cryjack 

I just did this workaround for myself since dealing/troubleshooting with the hwclock and time and TLS and the browser and ntpdaemon was becoming cumbersome, here is what I want you guys to try if you would like to prevent this from happening ever again...

Disable UZBL (browser) from verifying SSL connections.

get into the console/terminal (ctrl+alt+f1) or ssh into the raspberry pi and type:
sudo nano ./.config/uzbl/config

once in there, press ctrl+w and search for ssl, press enter
once it takes you to the line that says SSL related configuration, change the 1 in set ssl_verify = 1 to 0 and save by pressing ctrl + x , press y, press enter, then reboot by typing sudo reboot.

Pls let me know if this resolves the issue.

<i>disclaimer: disabling ssl verification is not recommended for common daily use of a browser, but this is just a workaround since the assets URL being added are trusted by you so there shouldn't be any worries regarding ssl connections.

iprz, 2019-02-08:
Hey there, I get the same error. I'm using pictures only along the weather and clock apps in a church lobby to show announcements and when error happens everything stops. 

*I got Screenly OSE from NOOBS, made sure timezone, local are correctly set and still got the tls msg in weather and clock websites

*To discard NOOBS problems, I formatted everything and loaded Screenly OSE directly from the image (not raspbian first) and I still got the error

*As last resort, I reformatted and setup the Pi with BalenaOS and for my surprise, it's been running for 12h without any issue with clock and weather.

I think deploying it if locally messes with something, sometimes I get the error in 1 minute after starting the assets or 5, 10 or 20 minutes later. But the Balena version so far has been working properly.

ealmonte32, 2019-02-08:
Guys I think I found the issue.. if you dont want to disable ssl verification from uzbl..
I was looking into the time sync error and since that is something that can cause TLS handshakes to go kaput, I believe that it has to do with ntp and systemd-timesyncd.service conflict because when you try to check status by running:
systemctl status systemd-timesyncd.service <i>(i was getting a condition failed error)</i>

<i>so then using this article as reference: https://feeding.cloud.geek.nz/posts/time-synchronization-with-ntp-and-systemd/</i>

Just removed ntp via purge with:
sudo apt purge ntp
then
sudo systemctl restart systemd-timesyncd.service
then
sudo reboot

and enabled ssl verify in uzbl config again and it has been working flawlessly since..

----
@iprz 
why cant you log in with ssh? did you put the SD card into computer and create a blank text file named ssh? you cant use ctrl+alt+f1 because ..? are you running experimental version or stable one ?
<i>did you read https://github.com/Screenly/screenly-ose/wiki - for ssh enabling instructions?</i>

why would you need to update config.txt ? this was never mentioned here, config.txt in boot and uzbl config are two different things..

ealmonte32, 2019-02-08:
I was looking into the time sync error and since that is something that can cause TLS handshakes to go kaput, I think that it has to do with ntp and systemd-timesyncd.service conflict because when you try to check status by running:
systemctl status systemd-timesyncd.service (i was getting a condition failed error)

<i>so then using this article as reference: https://feeding.cloud.geek.nz/posts/time-synchronization-with-ntp-and-systemd</i>

Just removed ntp via purge with:
sudo apt purge ntp
then
sudo systemctl restart systemd-timesyncd.service
then
sudo reboot

and enabled ssl verify in uzbl config again and it has been working since that..
if that doesn't help, I'll have to dig even deeper into more of the reasons why a TLS handshake could give the error besides system time... which frankly aren't many..

<i>dont forget to add an ntp server ip in /etc/systemd/timesyncd.conf if you dont already have one..</i>

@iprz
why cant you log in with ssh? did you put the SD card into computer and create a blank text file named ssh? you cant use ctrl+alt+f1 because ..? are you running experimental version or stable one ?
did you read https://github.com/Screenly/screenly-ose/wiki - for ssh enabling instructions?

why would you need to update config.txt ? this was never mentioned here, config.txt in boot and uzbl config are two different things..

ealmonte32, 2019-02-08:
I was looking more into this since some people may not want to disable uzbl ssl verify, and since issues with time sync is something that can cause TLS handshakes to go kaput, I thought that it could be due to system time + ntp service because when you try to check status by running:
systemctl status systemd-timesyncd.service , i was getting a condition failed error, ntp was inactive, and my ntp config was actually commented out as well.

<i>so then using this article as reference: https://feeding.cloud.geek.nz/posts/time-synchronization-with-ntp-and-systemd</i>

I did the following:
sudo apt purge ntp
then
sudo nano /etc/systemd/timesyncd.conf
then comment out the # in front of NTP= if you have it, and make it like this
[Time]
NTP=0.north-america.pool.ntp.org
<i>this is my preference, you make yours depending on your closest pool server from www.ntppool.org/en/</i>
ctrl+x to save, y, enter
then
sudo reboot

I also went ahead and re-enabled ssl verify in uzbl config again and it has been working fine since..
but.. i am still seeing the Insecure Warnings in journalctl with all https sites..

<i>
Feb 08 16:33:46 rp300we python[1009]: /usr/local/lib/python2.7/dist-packages/urllib3/connectionpool.py:847: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verifi
Feb 08 16:33:46 rp300we python[1009]:   InsecureRequestWarning)
Feb 08 16:33:46 rp300we python[1009]: Showing asset https://weather.srly.io/ (webpage)
and for boardclock, so we know it is not specific to any one https site..
Feb 08 16:34:17 rp300we python[1009]:   InsecureRequestWarning)
Feb 08 16:34:18 rp300we python[1009]: Showing asset https://boardclock.com/app/clock (webpage)
</i>

---
@iprz
why cant you log in with ssh? did you put the SD card into computer and create a blank text file named ssh? you cant use ctrl+alt+f1 because ..? are you running experimental version or stable one ?
did you read https://github.com/Screenly/screenly-ose/wiki - for ssh enabling instructions?

why would you need to update config.txt ? this was never mentioned here, config.txt in boot and uzbl config are two different things..

greenblueberry, 2019-02-13:
In school i've put youtube video rotate about 6h ago in that time Clock would be down. Now Screenly is working fine. without any tls problem.

ealmonte32, 2019-02-13:
@greenblueberry 
> In school i've put youtube video rotate about 6h ago in that time Clock would be down. Now Screenly is working fine. without any tls problem.

I am not sure what you meant by this.. are you saying you just changed the Asset to a YouTube URL instead of clock without doing anything else and now it is working fine? was it giving you TLS error on YouTube before or only clock URL?

greenblueberry, 2019-02-14:
Yes i changed youtube asset instead of clock. with clock url tls handshake was problem. but i dont know i'm just learning this raspberry and coding thing :D

ealmonte32, 2019-03-12:
@greenblueberry @iprz @TheDutchEdo @TheCazz @Cryjack 
A lot of research and a lot of testing is being done by me on this issue and I've narrowed it down to something with either of: python making the get request, uzbl, urllib3, ssl/tls, and ca certificates..

A workaround I thought about today since it always works, (as you also see with youtube URL), is to create a Google Sites and make a Page with just Inserting URL to it, and clear everything from it, header and all, and then under Insert > Embed > By URL, then in there you put the https://weather.srly.io/ and the boardclock one (they are both using the same CA certificate from comodo), but you will be always connecting to google.com so it should not give you TLS errors. When you finish just publish the page and put that as the Asset URL on your screenly Assets dashboard. Test and let me know, this should work.
You guys should add/create different pages for all the URLs that give you problems, and run them off Google Sites until a verified fix for TLS errors has been found.
That's my best suggestion right now.

Here is my sample page that I published: https://sites.google.com/view/weather-screenly/home

[image]

ealmonte32, 2019-03-28:
If this fails, I might give up :{
https://forums.screenly.io/t/crash-an-unexpected-tls-packet-was-received/36/26

The comment before this one, detailing the workaround is guaranteed to work for those who don't want to proceed with commenting out connection verification in urllib3.

th10gt, 2019-04-14:
I have the same problem with the weather app.
I'm showing with Screenly 6 pages from ThingSpeak and 3 satellite pictures from the internet. Adding the weather app allways crahed Screely after some few turns trough the 9 pages (plus weather app). But, to be honest, sometimes Screenly is crashing the same way without the weather app added. Time between such errors is 1 up to 2 days.

Seems Screenly isn't stable yet. 

And in addition and written already. When a web page can't be opened, Screenly shouldn't stop buut just show the next one...

Greetings from Germany!

ealmonte32, 2019-04-15:
@th10gt 
I have found a change to the uzbl configuration that has made screenly not crash or stop any longer for more than a week running..
Change your `./.config/uzbl/config-screenly` file to be this:
```
set show_status = 0
set on_event = request ON_EVENT
set cache_control = document_viewer
set fifo_dir   = /tmp
set socket_dir = /tmp

@on_event   LOAD_START    js alert=function(e){return true};confirm=function(e){return true};prompt=function(e){return true};
@on_event   LOAD_COMMIT   js alert=function(e){return true};confirm=function(e){return true};prompt=function(e){return true};
@on_event   LOAD_ERROR    exit
```

th10gt, 2019-04-15:
I fear, you have done more than that... After adding the lines and rebooting the RPI, Screenly starts, is performing some initializations but after that the screen is black...

th10gt, 2019-04-15:
Sorry, that was a stupid mistake. RPI for testing wasn't connected to the internet. Now it's running. Let see if Screenly is stablenow.

MANY THANKS!

th10gt, 2019-04-15:
... unfortunately I received this after less than 10 minutes... 
[image]

ealmonte32, 2019-04-15:
@th10gt 
So you edited your config-screenly file, restarted the Pi or screenly-viewer.service, and 10 minutes later got this? 
Did the next asset/URL come on after this? or did you have to click try again? the whole point of the `LOAD_ERROR exit` I was trying was to see if when the browser encounters an error, it "exits" and next asset/URL could load without user intervention, so this is why I want to make sure I get exact details from you.

If you did all this and asset did not go on next one by itself in the allowed time, I will then assume that the only difference regarding the relevant parts of this between your setup and mine would come down to the openssl and requests packages and certificates installed (updated cacert.pem extracted from mozilla root certs).
I'm a little busy lately but some of the other things i did to that Pi that doesn't get the error is remove `python-openssl` package (but leaving the one named `openssl`), I also installed `pip install requests[security]` prior, so troubleshooting and testing different things and comparing to what others did, led me to update the certificates and place the cacert.pem inside `/etc/ssl/certs/` and I also edited my openssl.cnf file..
I wish I knew exactly what my Pi has changed that we could apply to yours but my notes are all over the place with different Pi's and different things I did to different ones..

th10gt, 2019-04-15:
Hi,

first 4 x yes...

I had to cklick "Try again" and not the next but the weather came up. 

I think, I'll just skip the weather up in future. Or I need to ask a friend, being an RPI expert to support me with your hints. 

Thanks again!

th10gt, 2019-04-16:
Dear guys from Screenly,

as long as there seem to be problems to make a stable installation of Screenly, why you don't offer a stable running image of a (e.g.16GB) SD card, we all only need to update (e.g. WLAN credentials) only? In ZIP format it should have a size being too large for that...

vpetersson, 2019-04-16:
@th10gt - contributions are welcome. We release disk images at the end of every sprint. If you want something with commercial support, we offer [Screenly Pro](https://www.screenly.io/) (which is what pays for Screenly OSE).

ealmonte32, 2019-04-16:
> Dear guys from Screenly,
> 
> as long as there seem to be problems to make a stable installation of Screenly, why you don't offer a stable running image of a (e.g.16GB) SD card, we all only need to update (e.g. WLAN credentials) only? In ZIP format it should have a size being too large for that...

@th10gt your last line about ZIP format doesn't make sense.
And like vpetersson said, this is open source edition, you should be thankful there are volunteers even trying to make it stable and help you..

th10gt, 2019-04-16:
@vpetersson: Thanks for your answer and I appreciate that you're offering the free OSE version. But it's quite optimistic that someone having trouble with the stability of the free version, will buy the Pro version. Especially when the cost are monthly.

@ealmonte32 : Maybe my wording was missleading. What I've meant was, you can ZIP a 16 GB image down to some MB to offer it for download. I just shared an image this way with a colleague and it worked fine.

Greetings from Germany!

vpetersson, 2019-04-16:
@th10gt once upon a time Screenly Pro and OSE shared a code base. They do not anymore. They are completely different products. There's also a trial for Screenly Pro.

If you want to stick with OSE, that's great, but be mindful that it is a community project, so level expectations accordingly.

ealmonte32, 2019-04-17:
> Maybe my wording was missleading. What I've meant was, you can ZIP a 16 GB image down to some MB to offer it for download. I just shared an image this way with a colleague and it worked fine.

@th10gt 
Screenly already has pre-made images, see here:
https://github.com/Screenly/screenly-ose/releases

One thing I will advise is to setup the localization on the Pi first since that image defaults to en_GB and country and that messes with WiFi depending on your location, for example, I recently tested image 2018-11-23 and could not see a WiFi hotspot I created until I setup localization (all parts of it).

vpetersson, 2019-04-27:
I think this has been resolved now in the latest release. Closing out the issue. Comment if it is still present.

closed 2019-04-27 by vpetersson

th10gt, 2019-05-06:
Congrats to all! Since I've made a complete update, Screenly OSE is running since more than a week as a charm!