Screenly/Anthias #977: Backup not working for large amount of Assets

Cryjack, 2018-11-30:
#### Overview of the Issue

When starting a Backup for a large amount of Assets (>300) the Backup stops working and an empty Messagebox appears. The Backup is not created.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press "Settings"
* Press "Get Backup"
* Errormessage like in Screenshot appears

[image]

### Environment

* Raspberry Pi Hardware Version:
Raspberry Pi 3B+
* Screenly OSE Version:
latest dev (master-branch)

rusko124, 2018-11-30:
@Cryjack Also checked with assets more than 300, and this works for me.
Are you may not have enough free space?
To better understand the problem, can you show the logs, please?(`journalctl -u screenly-web`)

rusko124, 2018-11-30:
@Cryjack Got it, than can you provide information about free space on sd and size of directory "screenly_assets", please? (without screenly-backup.tar.gz)

Cryjack, 2018-11-30:
@rusko124 

"df -h" shows:
filesystem size used avail use% mountedon
/dev/root       7.3G  2.6G  4.5G  37% /

"du -hs /home/pi/screenly_assets" shows:
506M    /home/pi/screenly_assets

As far as I can see, no space problem.

rusko124, 2018-12-03:
Hi, @Cryjack ! Can you update and tell me if this has helped you?

Cryjack, 2018-12-04:
@rusko124 
I updated and checked the version.

The Webpage stays the same und does not change. But I can see, that the backup tar was created just fine in the filesystem. There is just no output on Webpage.

rusko124, 2018-12-04:
@Cryjack it's strange, the changes above should have helped.
An error without text still appears?

Cryjack, 2018-12-04:
@rusko124 

Yes, the same messagebox without content appears.

Cryjack, 2018-12-04:
The error on the website occurs while not finished creating the backupfile (still grows).

rusko124, 2018-12-04:
@Cryjack Hm, can you provide last commit hash, please?
You can also see the error in the web browser console(it should be associated with a timeout).
I can also offer you to open the Screenly web page in incognito mode,  there is a possibility that the JS script is cached.

Cryjack, 2018-12-04:
@rusko124 
No problem, here are the logs.

Here is the commit hash:
commit df48d332a2611ee2388825b87777da78fceb4e14
Merge: 40422ae b491563
Author: Viktor Petersson <vpetersson@users.noreply.github.com>
Date:   Mon Dec 3 10:34:43 2018 +0000

    Merge pull request #980 from rusko124/backup_timeout

    Set large timeout for backup

Here is the Nginx Errorlog:
2018/12/04 09:52:11 [error] 396#396: *244 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.44.5, server: , request: "POST /api/v1/backup HTTP/1.1", upstream: "http://127.0.0.1:8080/api/v1/backup", host: "172.16.0.73", referrer: "https://172.16.0.73/settings"

rusko124, 2018-12-04:
@Cryjack Well, Could you also tell me if you've rebooted your Raspberry after the upgrade, and can you provide **/etc/nginx/sites-enabled/screenly.conf** content, please?

Cryjack, 2018-12-04:
@rusko124 
yes, rebooted several times.

Content of screenly.conf:

upstream ose {
  server 127.0.0.1:8080;
}

upstream websocket {
  server 127.0.0.1:9999;
}

upstream wifi-connect {
  server 192.168.42.1:9090;
}

server {
  server_tokens off;
  listen 80 default_server;

  location / {
    return 301 https://$host$request_uri;
  }
}

server {
  server_tokens off;
  listen 443 default_server;

  ssl on;
  ssl_certificate /etc/ssl/screenly.crt;
  ssl_certificate_key /etc/ssl/screenly.key;

  location / {
    if (-f /home/pi/.screenly/initialized) {
      proxy_pass http://ose;
    }
    if (!-f /home/pi/.screenly/initialized) {
      proxy_pass http://wifi-connect;
    }

    client_max_body_size 4G;
    proxy_set_header X-Real-IP $remote_addr;
    proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
    proxy_set_header Host $http_host;
    proxy_set_header X-Forwarded-Protocol https;
  }

  location /static {
    proxy_pass http://ose/static;
  }

  location /ws/ {
    proxy_pass http://websocket;
    proxy_read_timeout 24d;
    chunked_transfer_encoding off;
    proxy_http_version 1.1;
    proxy_buffering   off;

    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection "Upgrade";
    proxy_set_header Host $http_host;
  }

  location /hotspot {
    allow 127.0.0.1;
    deny all;

    if (!-f /home/pi/.screenly/initialized) {
      root /tmp;
      rewrite ^ /hotspot.html break;
    }
  }
}

vpetersson, 2018-12-04:
@Cryjack Try upgrading. @rusko124 pushed some changes in #984.

Cryjack, 2018-12-04:
Just to let u know. I'd enabled SSL. Maybe this is something related to it?

vpetersson, 2018-12-04:
That could very well be it, @Cryjack. Could you disable that to see if it solves the problem? If so, we need to update [this file](https://github.com/Screenly/screenly-ose/blob/master/ansible/roles/ssl/files/nginx.conf).

rusko124, 2018-12-05:
@Cryjack Have you run the script `./bin/enable_ssl.sh` since the last update your Screenly?

Cryjack, 2018-12-05:
@rusko124 no. I havent run it after update. Should I do that?
@vpetersson Is there a way to disable it via script?

rusko124, 2018-12-07:
@Cryjack This script in your case will simply update your nginx configs.
Let me know if it has helped you.

Cryjack, 2019-01-03:
Hi together, 

sorry for the delay (christmas). A happy new year to all of you!

With the recent dev branch it is working. Thank you!

vpetersson, 2019-01-03:
Glad to hear that. Closing out the issue.

closed 2019-01-03 by vpetersson

ZhangXector, 2019-06-04:
I'm still experiencing this today. I don't feel like we have a large amount of assets, but we are seeing the same issues. I had a post about this in the Screenly OSE forums, but can't seem to find it (might have been archived?). I'm attaching the error message, system information, and the log file. Hopefully this can help shed some light on this because it's one of our most used features.

System information:

Load Average | Free Space | Uptime | Monitor Info | Display Power | Raspberry Model | Screenly Version | MAC Address
-- | -- | -- | -- | -- | -- | -- | --
0.84 | 26G | 0 days and 1.04 hours | state 0x12000a [HDMI DMT (39) RGB full 16:9], 1360x768 @ 60.00Hz, progressive | On | Model 3B Revision: 1.2 Ram: 1 GB Embest | development@76c881c |  

Chrome console log:
```
Failed to load resource: the server responded with a status of 504 (Gateway Time-out)
VM279:1 Uncaught SyntaxError: Unexpected token < in JSON at position 0
    at JSON.parse (<anonymous>)
    at Function.parseJSON (jquery-1.9.1.min.js?q=1549053199:3)
    at HTMLInputElement.fail (settings.js?q=1559683079:71)
    at e.<computed>.<computed>._trigger (jquery-ui-1.10.1.custom.min.js?q=1549053199:6)
    at e.<computed>.<computed>._onFail (jquery.fileupload.js?q=1549053199:673)
    at e.<computed>.<computed>._onFail (jquery-ui-1.10.1.custom.min.js?q=1549053199:6)
    at Object.<anonymous> (jquery.fileupload.js?q=1549053199:715)
    at c (jquery-1.9.1.min.js?q=1549053199:3)
    at Object.fireWith [as rejectWith] (jquery-1.9.1.min.js?q=1549053199:3)
    at k (jquery-1.9.1.min.js?q=1549053199:5)
parseJSON @ jquery-1.9.1.min.js?q=1549053199:3
fail @ settings.js?q=1559683079:71
_trigger @ jquery-ui-1.10.1.custom.min.js?q=1549053199:6
_onFail @ jquery.fileupload.js?q=1549053199:673
(anonymous) @ jquery-ui-1.10.1.custom.min.js?q=1549053199:6
(anonymous) @ jquery.fileupload.js?q=1549053199:715
c @ jquery-1.9.1.min.js?q=1549053199:3
fireWith @ jquery-1.9.1.min.js?q=1549053199:3
k @ jquery-1.9.1.min.js?q=1549053199:5
r @ jquery-1.9.1.min.js?q=1549053199:5
```

[screenly_output.zip](https://github.com/Screenly/screenly-ose/files/3254731/screenly_output.zip)

rusko124, 2019-06-05:
Hi @ZhangXector !
I cannot reproduce it in the last commit.
Could you provide `/etc/nginx/sites-enabled/screenly.conf` content, please?

ZhangXector, 2019-06-05:
I looked in that directory and didn't see anything called that. The only file in that directory (and in `sites-available`) is the `default` file.

Because this Raspberry Pi has had Screenly OSE manually installed from source and updated several times, I'm taking the steps to wipe it and install the latest stable version from the zipped image (Sprint 9). I'll report back to see if this makes any difference.

ZhangXector, 2019-06-05:
After reinstalling with the Screenly sprint 9 image, we are still seeing the problem with backing up from one raspberry pi and restoring to another. We get the pink empty warning (I assume that's the timeout error). After reloading the page, all the assets appear to be there (which is strange). However, none of the videos are running even after a reboot. All I see is the darkened Screenly page. Still no sign of a screenly.conf in the nginx sites-available either. . .

rusko124, 2019-06-06:
@ZhangXector It's strange. How much long are you waiting for the error?

ZhangXector, 2019-06-06:
After the upload of the backup file gets to 100%, it usually takes 2-3 minutes for the error to appear. What's weird is that it looks like it's creating the assets, but then because of the timeout they don't seem to finalize, so they never actually play.

rusko124, 2019-06-10:
@ZhangXector Could you also specify the size of the backup? I will try to create a similar one and reproduce it.

ZhangXector, 2019-06-10:
@rusko124 446.4 MB