Screenly/Anthias #979: Doesnt load nearpod website?

ealmonte32, 2018-11-30:
#### Overview of the Issue

Just trying to load any of the following links but the uzbl browser doesnt seem like the content:
https://nearpod.com/s/digital-citizenship/kindergarten/going-places-safely-lite-L31462922?_branch_match_id=597132583168613097

https://app.nearpod.com/presentation?pin=3827FAB236FC71699FBCA3470419B471-1

#### Reproduction Steps

Steps to reproduce this issue:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen
Other URLs load fine like google slides, time.gov, worldclock, google calendar, etc..

### Environment

* Raspberry Pi Hardware Version: pi 3 model b
* Screenly OSE Version: latest developer version upgraded manually by running script

pi@rp300we:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian

pi@rp300we:~ $ uname -a
Linux rp300we 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

vpetersson, 2018-12-01:
@ealmonte32 Could you try the experimental branch? 

That said, those pages do look pretty heavy. Pages with that much JavaScript (in particular animations etc) are known to not work well.

rusko124, 2018-12-04:
Hi @ealmonte32.
I confirm that nearpod's pages are not displayed in the experimental branch.
But I also checked the developer version, and it shows pages.

Yes, you are right. Screenly has production, developer and experimental branches.
The experimental branch uses Docker containers and QT technology instead of uzbl.

ealmonte32, 2018-12-04:
Hmm.. I see..

So will developer branch be better in the sense that it sort of supports more content/media?

If so, how do i install the developer version from terminal? (I originally installed screenly ose via image, but then upgraded via bash script and selected yes to experimental branch)

Will it delete all the current screenly assets? (not a big deal i just want to document everything before proceeding)

Thx

ealmonte32, 2018-12-11:
Sorry I closed it too soon.. i ran the upgrade, upgraded to developer mode, and the assets are now empty, but the assets in folder screenly_assets seem to be there? how do i know what assets are on the screenly_assets folder if they dont show on the browser..

screenshots below:

[image]

[image]

vpetersson, 2018-12-11:
Could you try `echo "SELECT * from assets" | sqlite3 ~/.screenly/screenly.db` (assuming i remember the syntax right by heart ;)

vpetersson, 2018-12-16:
> Good memory, but nothing happens after that command, the assets page is still empty..

That's expected. That simply dumps the content of the database.

Are the assets restored if you switch to the production branch?

ealmonte32, 2019-01-03:
> > Good memory, but nothing happens after that command, the assets page is still empty..
> 
> That's expected. That simply dumps the content of the database.
> 
> Are the assets restored if you switch to the production branch?

No they are not... 

I think you guys should note that down when someone upgrades from one branch to other the assets are retained on the system but they dont start or the screenly viewer doesnt recognize them or accept the assets when dumped back onto database..

Since I couldn't spend more time trying to figure out how to get it to work, I just downloaded the new screenly image, but now when i try to join the wifi network it creates it gives me incorrect password..? like literally I am typing it 100% correct.. I tried from macbook, from ipad, there's definitely something wrong, am i the only one with this issue on newest released image?

[image]

ealmonte32, 2019-01-03:
So i remembered that you guys are in UK and I am in US, and i realized this when I was trying to check some of the settings in ./screenly and my keyboard kept putting weird characters, so I went ahead and changed localization settings via raspi-config, then i went ahead and edited the start resin wifi script and deleted the password generator part and made it open network, and now am finally on it, but uploading the backup I made did not backup any assets, and now when i try to add an asset it says the following:

[image]

vpetersson, 2019-01-04:
> So i remembered that you guys are in UK and I am in US, and i realized this when I was trying to check some of the settings in ./screenly and my keyboard kept putting weird characters

That's strange. While I'm myself in the UK, we have a global team and the disk image is actually using `en_us` locale. 

@rusko124 please see if you can reproduce this.

ealmonte32, 2019-01-04:
> That's strange. While I'm myself in the UK, we have a global team and the disk image is actually using `en_us` locale.

> @rusko124 please see if you can reproduce this.

Well, I can tell you right now I downloaded the screenly image literally 2 days ago and the locale was set to en_GB.. 

I have to prepare 4 screenly TVs and flashed 4 drives with the latest image and just to prove it to you since I already changed locale on other, I booted up this new one , here are pictures:

[image]

[image]

@vpetersson So i should not try to upload backup from previous version experimental branch in order to avoid getting this skip_check error on assets? Does that backup include assets or only configuration settings in the screenly.conf file? I deleted the assets from the screenly_assets folder to see if that helped resetting anything and i still get the skip check error on anything i try to upload.. how can i reset the whole screenly config in order to not have skip error? re-running the bash script as if i was reinstalling the screenly from scratch?

rusko124, 2019-01-07:
@vpetersson Yes, the raspi-config is set to `en_GB` locale.
[image]

@ealmonte32 
Yes, backup include assets.
If you want to get resolve this error without updating Screenly-OSE, I can advise you to run a python script `python /home/pi/screenly/bin/migrate.py`
This will add the required columns to db

vpetersson, 2019-01-07:
> @vpetersson Yes, the raspi-config is set to en_GB locale.

Strange. The old disk image instructions set it to `en_US.utf8`. Did we lose that with the new disk image process?

rusko124, 2019-01-07:
@vpetersson 
This instruction part of install.sh.
But it doesn't executed when image build.
So you are right, we need to make edits for https://github.com/Screenly/pi-gen

vpetersson, 2019-01-07:
@rusko124 could you please create a ticket for that?

ealmonte32, 2019-01-07:
>@rusko124 
>Yes, backup include assets.
>If you want to get resolve this error without updating Screenly-OSE, I can advise you to run a python >script python /home/pi/screenly/bin/migrate.py
>This will add the required columns to db

Thank you this worked, this fixed the skip_check on assets issue after backup restore..
Maybe you guys can add that simple command to the restore process, like if someone restores from an older version or experimental branch to developer branch or whatever, that after the restore process finishes that it runs the migrate python script automatically just as a fail-safe to make sure it restores completely... just a suggestion that should probably avoid any future issues with restoring.. unless migrate.py script can cause other isssues...

Thanks anyway.

closed 2019-01-07 by ealmonte32