Screenly/Anthias #982: weak signal then doesn't reconnect

ealmonte32, 2018-12-03:
#### Overview of the Issue

I read this post:
https://support.screenly.io/hc/en-us/community/posts/360001834573-Screenly-OSE-does-not-seem-to-reconnect-to-WiFi-

and this one:
https://support.screenly.io/hc/en-us/community/posts/360007982734-Changing-to-new-WiFi-v0-15-1-Sprint-7-

But I still dont think the question is answered, basically, without turning off the pi, it seems the screenly-ose pi doesn't reconnect to wifi after for some reason it disconnects from the access point.
I am attaching the screenshot that shows the pi at -58dBm (82% signal) and yet it seems to disconnect from the wifi with weak signal.. and when i move the pi "closer" to the AP, it doesn't reconnect until i manually reboot.. this doesn't make sense because if rebooting but leaving in same location means the signal was the same prior to rebooting so this means the wifi manager was not reconnecting properly.. 
I am trying to get familiar with the different network manages such as wicd, but I dont want to mess with the screenly built-in wifi manager unless someone has found that replacing the one it comes with fixes these issues..

Any workarounds or things to test i will be glad to test..

#### Reproduction Steps

While screenly-ose pi has wifi connection, reboot the access point, see if pi automatically reconnects to wifi without rebooting the pi.

### Environment

Raspberry Pi Hardware Version: pi 3 model b
Screenly OSE Version: latest experimental branch version upgraded by running script about 1 week ago

pi@rp300we:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian

pi@rp300we:~ $ uname -a
Linux rp300we 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

[image]

vpetersson, 2018-12-04:
@ealmonte32 Strange. We don't really do anything custom at all as far as WiFi goes. We just use [WiFi Connect](https://github.com/balena-io/wifi-connect) from Balena to simplify the bootstrapping process. Other than that, it's all just Network Manager. Perhaps you can try tweaking the NM settings to see if that helps.

ealmonte32, 2018-12-11:
@vpetersson 
Thanks for info, yeah I knew you guys use the balena/resin bootstrap for wifi connect..
I just added the iface wlan0 inet dhcp etc to my network interfaces file and it seems to reconnect now, it was empty before.. oh well.

closed 2018-12-11 by ealmonte32

trustees629, 2019-03-30:
@ealmonte32 , I'm sorry to be reopening this issue but I cannot find another way to ask for help. I have one system that keeps dropping wifi connection and will stay offline until I reboot. I saw where you added iface wlan0 inet dhcp etc to your network interfaces file. My question is, where is this file? I have looked all through my SD card and I can't find it! I've literally searched Google for days trying to figure this out. Also, I cannot figure out how to install wifi connect. It's driving me crazy!!

ealmonte32, 2019-04-01:
> @ealmonte32 , I'm sorry to be reopening this issue but I cannot find another way to ask for help. I have one system that keeps dropping wifi connection and will stay offline until I reboot. I saw [...]

@trustees629 
The interfaces file is in `/etc/network/`

But I just want to say, I would still experience those reconnect issues doing it that way, I just did not come back here and post about it but instead did my own research to figure out how the raspberry pi handles the whole wifi networking system. So I stopped doing the wifi through that file because sometimes I would see in the logs that the dhcp conflicts with the wpa_supplicant one. _(there are multiple dhcp clients on a base install of raspbian, dhcpcd5 and isc-dhcp-client, and you can search online as to why, one quick reference: [link](https://raspberrypi.stackexchange.com/questions/39785/differences-between-etc-dhcpcd-conf-and-etc-network-interfaces)

Assuming you have not changed any settings besides set the wifi up via `raspi-config`, it creates or edits the file /etc/wpa_supplicant/wpa_supplicant.conf and sets it to run at start and does all the backend stuff for you.

Then I installed wicd-curses just so that it uses the reconnect feature it has, and that was working fine for a while, but now i want to try a program called connman: 
https://packages.debian.org/stretch/connman

That should help with reconnects because as far as I've read, the wpa_supplicant does not really have a built-in function for reassociation/reconnects: [wpa_supplicant ctrl iface](https://w1.fi/wpa_supplicant/devel/ctrl_iface_page.html) so it allows other programs to access the current status or state of the connection, and then it is supposed to reconnect for you if the status or state becomes disconnected.
Give either of those programs a try, and do not edit your /etc/network/interfaces file, leave alone so that there are least possible conflicting things running in background.

ealmonte32, 2019-04-02:
@trustees629 
Do not install connman on a wifi only Pi unless you have access to it physically or else you will lose connectivity.. just tested and it kills the wifi and starts other servers and I had to go and physically plug in a keyb and mouse and do rfkill unblock wifi and restart all wpa supplicant and networking services and reboot before it started working again.. 
So unless you first practice and read how to configure the network on debian using connman.conf file and the program, dont try this.

trustees629, 2019-04-02:
Thanks for the update. I'm running through Balena so I don't believe I have all those other files you referred to anyway. Seems to be working fine now as I moved my Pi to a better spot. However, I never did figure out how to install wifi-connect through Balena.

ealmonte32, 2019-04-02:
You're running screenly through balena? are you doing the raspberry pi 3 or pi zero?

miragefoto, 2019-10-14:
I am having the same problems.   My pi's boot before the router when the power goes out, so on launch they do not see the AP.  It seems they never try to reconnect even though it is available as few seconds later.

Has anyone figured out a good solution for this?

vpetersson, 2019-10-14:
There's no built-in feature for this, but it can be solved fairly easily.

You can create a bash script like this:

```
#!/bin/bash

if [ $(cat /sys/class/net/wlan0/operstate) == "down" ]; then
    service networking restart
fi
```

And then you can run it every hour or so using a cronjob. 

This will only work on Raspbian and not Balena unfortunately.