Screenly/Anthias #990: idea - time of day window

dodds-steven, 2018-12-10:
I am new to Screenly and a thought came to me today as I was configuring the first video to playback.
I didn't see an option in the settings, but thought... Hmmm it could be handy to have a way to say only play this asset between these hours and maybe even on these days only.  
Examples may be; 

- after hours messages
- mix up what videos are shown through out the day, such as play this one only in the morning, and another only in the afternoon.

I like this system so far, it's nice and I see it getting better. 
thanks!

vpetersson, 2018-12-10:
This is a feature that we already support in Screenly Pro. Don't think we'll port this functionality unfortunately (unless someone else will do the work).

dodds-steven, 2018-12-12:
Oh, I didn’t see that listed in the comparison I read on the website.
Ok...

closed 2019-02-17 by stale[bot]