Screenly/Anthias #991: Add the sophisticated playlist editor

achikas, 2018-12-12:
It would be very user-friendly to schedule slides for recurring days.
I read that the pro Edition has the option to set rules for weekdays in addition to timeslots.
https://support.screenly.io/hc/en-us/articles/217874263-Using-Screenly-s-sophisticated-playlist-editor

I really like to see that feature in the OSE Edition.
Can it be managed?

vpetersson, 2018-12-12:
Hi @achikas - this is not something that we have planned for OSE.

closed 2018-12-12 by vpetersson

achikas, 2018-12-12:
these are not good news.
hope you will consider to think about.
until then I need to use my website to deliver the content for recurring weekdays.

vpetersson, 2018-12-12:
It's pretty simple. Screenly Pro is what pays for our development cost of Screenly OSE. As such, we will keep some features out of OSE.