Screenly/Anthias #992: Cannot get to terminal using Ctrl+Alt+F1

Powers41, 2018-12-14:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

A paragraph or two about the issue you're experiencing.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version:
* Screenly OSE Version:

Powers41, 2018-12-14:
Trying to get to the terminal to configure wifi the ctrl+alt+f1 does not access terminal.

Raspberry Pi 3 b vi2
OSE version 2 fresh install

vpetersson, 2018-12-15:
Just download the latest disk image from [here](https://github.com/Screenly/screenly-ose/releases). There's no need for configuring wifi using the terminal as there is a built-in functionality for setting up WiFi.

closed 2018-12-15 by vpetersson