Screenly/Anthias #993: Database issues during interrupted update

loleg, 2018-12-17:
The Raspberry Pi froze during the system upgrade process launched through the OSE update script. After a reboot, Screenly's new version loaded but no Assets were visible, and adding a new asset produced `Server Error: no such column: skip_asset_check`. I assume that somehow the database update didn't happen.

My suggestion is to at least change the messaging of the option to say that it is really not recommended. Or leave it out altogether. An improvement would be to ensure that Screenly is completely updated before the system upgrade starts.

#### Reproduction Steps

* From the [Pi's console](https://github.com/Screenly/screenly-ose/wiki#console-access), run the install script as [documented here](https://github.com/Screenly/screenly-ose/wiki#updating-screenly)
* Select "Yes" to upgrade the system as well.
* Restart the Pi mid-way (I don't recommend doing this, it may damage your system!)
* Try using the Screenly web interface

### Environment

* Raspberry Pi Hardware Version: Zero
* Screenly OSE Version: latest... [I guess](https://github.com/Screenly/screenly-ose/issues/120)

vpetersson, 2018-12-17:
@loleg Not very surprising and hard to protect against (we can't prevent someone from pulling the power half-way through). That said, we are working on improving the upgrade flow in #987. As a short-term remedy, simply re-running the upgrade script should do the trick.

loleg, 2018-12-17:
I did re-run the upgrade script, and it did not fix the issue, even though I did not see an error in the migration step. So I deleted the database and start afresh. I'm now not even sure that the root cause was the frozen system upgrade. I'll see if I can reproduce the problem on another device, but if you think this issue is too much of an outlier, you can close it.

rusko124, 2018-12-18:
Hi @loleg !
Can you try this command?
`python /home/pi/screenly/bin/migrate.py`

closed 2019-02-25 by stale[bot]