Screenly/Anthias #995: Assets not showing

DaveEps, 2018-12-18:
Hi i am new on here and fairly new to Screenly !
i had screenly up and running on an old Raspberry pi for a few months 
and it worked so well with my website , i thought i would upgrade to the latest Pi ,
as the old Pi was very slow .
my main asset is a link to my gallery on my website , which just shows images to any customers coming in.
but for some reason it will not load on the new one ?
so i have tried numerous website testing them and some load and some dont !
for example , Facebook home page worked but,
facebook with the extension to my page wouldnt load ?

sorry if this is a duplicate of an other issue but i have searched for other similar problems but cant find anyone with the same .
i have attached a log of when i turn the asset on .
be much appreciated if someone could help .
thanks in advance 
[screenly.pdf](https://github.com/Screenly/screenly-ose/files/2689866/screenly.pdf)

vpetersson, 2018-12-18:
Hard to say, but please fill out the steps for reproducing the issue as per the template.

DaveEps, 2018-12-19:
Thanks for the reply back ,
also sorry i deleted that part thinking i didnt need it !!

i have Raspberry Pi 3 Model B+ which i installed the newest version of screenly from NOOBS_v2_8_2 .

For example to reproduce what i have done .

*added asset - webpage = https://www.facebook.com/
*turn on asset - waited for like a minute
*facebook homepage shows up on screen (works perfect)
*turn off asset

*added asset - webpage = https://www.facebook.com/EPSbn9/
*turn on asset - waited for like a minute
*screen flashes white over and over again like its almost not connecting to the website / or website is taking to long to respond .
*turn off asset- stops flashing and goes back to blank page with screenly logo.

to note : https://www.instagram.com/?hl=en - worked but when i added my instagram page 
https://www.instagram.com/epsbn9/?hl=en it didnt work 

this also doesnt work with my website http://expandedpoly.co.uk/ - 
but i have just found out i have a problem with my site ! which is getting fixed as we speak .

i have also added several other webpages to test which worked fine ; 
https://www.ebay.co.uk/
http://www.xmasclock.com/
https://www.instagram.com/?hl=en
https://github.com/Screenly/screenly-ose/issues

if you need any other info that i might of missed let me know .
and thanks in advance for the help .
Dave

DaveEps, 2018-12-20:
Just to note : during installation of screenly ,
the raspberry Pi turned off twice due to lack of power ! i also had a flashing lighting bolt sign in the corner of the screen .
i noticed i was using an old cable, which ran my old Pi but worked fine , looking into it the output was 0.2 Amps , i bought an official raspberry pi power lead with 2.5 Amps and now all is fine !
with this in mind i started to think that this maybe a missing file or two due to shutting down on install maybe the problem ??
also noticing that there is a newer version of NOOBS too .
so i formatted the SD and started afresh again !
downloaded NOOBS v3.0 and installed screenly from the install menu .

But again same things are happening !
some webpages are working others are not .
it almost seems like the websites which are not working take longer to load and screenly has a look dont get a response quick enough so doesnt connect , if you get what i mean ?

take for instance : facebook home page (where you login) works fine load time is very fast .
but try and load a page from facebook, it takes longer due to all the other elements on the page and this doesnt work on my screenly ?
i maybe barking up the wrong tree but thought ill add as it seemed to point to load time on my tests .

vpetersson, 2018-12-20:
A few things:

 * The PSU issue is expected. We see *a lot* of issues with this. That's why we only recommend using the official PSU
 * There could still be issues with the SD card. I'd try with either replacing the SD card or using [SD Formatter](https://www.sdcard.org/downloads/formatter_4/) and do an 'Overwrite format'
 * The website issue isn't all that surprising. The Facebook page is **heavy**. It is not the type of content Screenly is designed for. It works best with tailored content that isn't using too much JavaScript or fancy CSS.

DaveEps, 2018-12-20:
thank you for getting back to me .
the SD card is brand new and i used the SD formatter to format and reinstall .
i only used the Facebook website as a test , so i wouldn't be using that site or anything similar at all .
for future, i planned to create a separate page hidden on my server which would just be a slideshow of pictures that i want showing on my screen .
but for now i just wanted to run my gallery on my website , which it did run fine and i had running for everyday for around 4 months on the Old raspberry Pi .

But still confused ! 
why it worked fine before on a model B 1.2 and now doesn't , even though it is now on a faster device - bigger SD card and also i have ran an Ethernet cable and hard wired it to the internet !!
any ideas would be much appreciated that i could try  :-)

vpetersson, 2018-12-20:
>  i planned to create a separate page hidden on my server which would just be a slideshow of pictures that i want showing on my screen

Is there any particular reason for not using the built-in functionality for this? It would provide you with much better performance and stability.

> why it worked fine before on a model B 1.2 and now doesn't , even though it is now on a faster device - bigger SD card and also i have ran an Ethernet cable and hard wired it to the internet 

It's really hard to say. Do you know what version of Screenly the old device ran? If so we might be able to compare and contrast and see if we can find out.

DaveEps, 2018-12-20:
the only reason i would not use the in-built functionality is that i could create effects on a webpage that would go from one picture to the next . would be a good function though ! to have on screenly if we could fade/slide in the next image/webpage ?
i will work on adding some images for the time being .

i installed NOOBS v2.8 on the old Pi and NOOBS v3.0 on the new Pi 
but not sure if there is a difference to screenly in them , as if i remember rightly i just chose screenly as an OS and it automatically downloads it .

i searched the web for a decent signage software and have used multiple devices including apple tv / google chromecast which both would not work alone and needed a pc to set up every morning ! which was a pain !! i really like the concept of screenly and hopefully i can get it working .

vpetersson, 2019-01-02:
The version of NOOBS shouldn't matter. @rusko124 can you check if you're able to reproduce this?

DaveEps, 2019-01-07:
@vpetersson @rusko124 Great news thank you to both of you for looking into this problem for me .
absolutely love Screenly and was a little annoying it wouldn't show my website .
now i can go back to my older plan and maybe even add another screen !

but for now i have one last Question ?
how do i upgrade ?? 
lol sorry i did say i was new !

ealmonte32, 2019-01-07:
> @vpetersson @rusko124 Great news thank you to both of you for looking into this problem for me .
> absolutely love Screenly and was a little annoying it wouldn't show my website .
> now i can go back to [...]

Run the script rusko suggested, if you need to get to terminal press ctrl+alt+f1 on the keyboard and when in terminal log in, then just type what he wrote on console:
./bin/run_upgrade.sh

DaveEps, 2019-01-07:
@ealmonte32 i did think i had to do this but thought i would double check before i added it somewhere i should not then having to format and start again lol
thanks for you help :-)

ealmonte32, 2019-01-07:
> @ealmonte32 i did think i had to do this but thought i would double check before i added it somewhere i should not then having to format and start again lol
> thanks for you help :-)

lol, i hear that.. double checking is always a good thing

DaveEps, 2019-01-07:
@ealmonte32 just tried this and comes back with
`-bash: ./bin/run_upgrade.sh: No such file or directory`
also tried using sudo nano too ?

ealmonte32, 2019-01-07:
> @ealmonte32 just tried this and comes back with
> `-bash: ./bin/run_upgrade.sh: No such file or directory`
> also tried using sudo nano too ?

nano is a text editor.. dont worry you just need to find the right directory..
rusko was just giving you the short way to get to the directory, you should be able to find it here:

/home/pi/screenly/bin/run_upgrade.sh

DaveEps, 2019-01-08:
@ealmonte32 thank you for this :-)
i have just ran the script this morning .
all is now back up and running and reading my website again !
thank you again to @vpetersson @rusko124 @ealmonte32 for all your help.
think i would of gave up if it wasnt for you all !!

also i think i am going to do what @vpetersson said to do and im going to use the website and also upload pictures as well for hopefully a smoother operation .
thank you 
Dave

Kubilai12, 2019-01-08:
Thanks so much i had same this afternoon with google slide link i upgrade and it s ok

I try to configure otherthings now but when i do CTRL+ALT+F1 i don t go on console mode does the upgrade change this ?

ealmonte32, 2019-01-08:
> @ealmonte32 thank you for this :-)
> i have just ran the script this morning .
> all is now back up and running and reading my website again !
> thank you again to @vpetersson @rusko124 @ealmonte32 for [...]

Glad its working Dave.. maybe you can close this issue so that it doesn't show open now.
Thanks.

ealmonte32, 2019-01-08:
> Thanks so much i had same this afternoon with google slide link i upgrade and it s ok
> 
> I try to configure otherthings now but when i do CTRL+ALT+F1 i don t go on console mode does the upgrade change this ?

@rusko124 or @vpetersson could correct me if im wrong but as far as i've been told, experimental and developer branch do not allow ctrl+alt+f1, can you ssh into the device? if not, you can just place a blank file into the boot directory of the SD card and name it ssh even if it has .txt as extension, and upon boot the rpi would have ssh enabled..

vpetersson, 2019-01-08:
@ealmonte32 - yes you're correct. The experimental branch takes over the screen so you will not bel able to use a keyboard to gain access to the shell (at least not visibly).

Kubilai12, 2019-01-09:
> if not, you can just place a blank file into the boot directory of the SD card and name it ssh even if it has .txt as extension, and upon boot the rpi would have ssh enabled..

Thanks for this tips

closed 2019-01-09 by DaveEps