Screenly/Anthias #996: No hotspot when using balenaCloud

astrodad, 2018-12-19:
#### Overview of the Issue

I just configured a brand new Application and two devices in balenaCloud, all on ethernet. I disconnected the ethernet and rebooted the devices, expecting Screenly to create the hotspot so that I could configure the wifi. No hotspot appears and Screenly cannot connect. 

### Environment

* Raspberry Pi Hardware Version: Raspberry 3 B+
* Screenly OSE Version: built from Master branch today via balenaCloud

ealmonte32, 2019-01-03:
I've seen this and experienced it before, this is just one of many possible ways to reset the wifi connection screen..

get into terminal by pressing ctrl+alt+f1 , and once there run:
ls ~/.screenly/
and see if you have a file called wifi_set

if so, delete it by running:
rm -rf ~/.screenly/wifi_set

rusko124, 2019-01-07:
Hi @astrodad !
In the balenaCloud image settings, you can specify your access point.
Currently, Screenly hotspot is not supported in balenaCloud

[image]

ealmonte32, 2019-01-16:
@astrodad can you close this issue if it was resolved by what rusko suggested?

ealmonte32, 2019-02-23:
@vpetersson - can we close this since user has not replied?

astrodad, 2019-02-23:
Thanks for the suggestion and the info.

closed 2019-02-23 by astrodad