Screenly/Anthias #998: Backup Authentication Bypass

AngelG-543, 2018-12-23:
When a screenly instance is setup and running, there is normally authentication if the user chooses to enable it that guards the various functions. A recent discovery has led to being able for an unauthenticated user to bypass the authentication mechanism and retrieve a copy of the backup file.
screenly setup:
Authentication enabled
no ssl enabled
--------------------
It seems that the link is not active when authentication is disabled. The steps to reproduce are:
setup authentication
save settings
navigate to <host>/static_with_mime/screenly-backup.tar.gz?mime=application/x-tgz using a different browser, machine, or clearing browser data
download begins
[image]

----------------------
This was also tested using another raspberry pi that had screenly installed on top of raspbian-lite. same results. The way this could be fixed could be to just change the permissions available for the download, or to require authentication when going directly to the link.
--------------------------------------------------------------------
The configuration file contains all types of good information such as a SHA-256 password hash and a Cleartext username

vpetersson, 2018-12-23:
@AngelG-543 Good find. Frankly speaking, I wouldn't recommend using the built-in authentication as it is still pretty rough and should be considered work-in-progress. If you want to secure the device properly, I suggest using basic auth in Nginx instead. That said, we should definitely fix this.

@rusko124 could you try reproducing this?

rusko124, 2018-12-24:
Confirm the issue above.
Created a PR to resolve this problem.

closed 2019-03-01 by stale[bot]

vpetersson, 2019-03-01:
@rusko124 Is this fully solved?

rusko124, 2019-03-01:
@vpetersson I think yes, we can close it.

vpetersson, 2019-03-01:
@rusko124 can you  just double check to ensure you cannot reproduce it?

rusko124, 2019-03-01:
> @rusko124 can you just double check to ensure you cannot reproduce it?

@vpetersson 
[image]