Screenly/Anthias #1002: Can't activate assets

TheCazz, 2018-12-27:
Hi
I running the latest Screenly (I did upgrade because I did got same problem and was thinking it maybe fix the problem if I do a upgrade) and have raspberry pi 3

Today when I was going to change the play list I can inactive the assets (OFF) but I can't activate any assets that I have on the raspberry pi. No pic or movie wont to be activate.

It just show "ON" but they does not move up to the activate assets??

the log just say

Dec 27 06:48:38 PiTVGISANG1112 python[567]: Last update: 2018-12-27 06:39:21.590322
--
Dec 27 06:48:38 PiTVGISANG1112 python[567]: get_next_asset
Dec 27 06:48:38 PiTVGISANG1112 python[567]: refresh_playlist
Dec 27 06:48:38 PiTVGISANG1112 python[567]: refresh: counter: (0) deadline (2018-11-23 12:49:00) timecur (2018-12-27 06:48:38.136236)
Dec 27 06:48:38 PiTVGISANG1112 python[567]: update_playlist
Dec 27 06:48:38 PiTVGISANG1112 python[567]: Generating asset-list...
Dec 27 06:48:38 PiTVGISANG1112 python[567]: generate_asset_list deadline: 2018-11-23 12:49:00
Dec 27 06:48:38 PiTVGISANG1112 python[567]: get_next_asset after refresh
Dec 27 06:48:38 PiTVGISANG1112 python[567]: Playlist is empty. Sleeping for 5 seconds
Dec 27 06:48:38 PiTVGISANG1112 python[567]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Dec 27 06:48:43 PiTVGISANG1112 python[567]: Last update: 2018-12-27 06:39:21.590322
Dec 27 06:48:43 PiTVGISANG1112 python[567]: get_next_asset
Dec 27 06:48:43 PiTVGISANG1112 python[567]: refresh_playlist
Dec 27 06:48:43 PiTVGISANG1112 python[567]: refresh: counter: (0) deadline (2018-11-23 12:49:00) timecur (2018-12-27 06:48:43.242390)
Dec 27 06:48:43 PiTVGISANG1112 python[567]: update_playlist
Dec 27 06:48:43 PiTVGISANG1112 python[567]: Generating asset-list...
Dec 27 06:48:43 PiTVGISANG1112 python[567]: generate_asset_list deadline: 2018-11-23 12:49:00
Dec 27 06:48:43 PiTVGISANG1112 python[567]: get_next_asset after refresh
Dec 27 06:48:43 PiTVGISANG1112 python[567]: Playlist is empty. Sleeping for 5 seconds
Dec 27 06:48:43 PiTVGISANG1112 python[567]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.

rusko124, 2018-12-27:
Hi @TheCazz !
I just tried to reproduce these issue, but everything works well for me.
My steps: I installed the latest image and then updated.

Can you provide more information, please? You also make a backup of your assets and repeat the steps I wrote above.

TheCazz, 2018-12-27:
Right now it works nice after I have reinstall everything. Just hope that I do not need to do that with the other screenly

TheCazz, 2019-01-10:
Have same with another one even after I have upgrade it

Jan 10 15:42:04 PiTVGISAND115 python[585]: Generating asset-list...
--
Jan 10 15:42:04 PiTVGISAND115 python[585]: Playlist is empty. Sleeping for 5 seconds
Jan 10 15:42:09 PiTVGISAND115 python[585]: Generating asset-list...
Jan 10 15:42:09 PiTVGISAND115 python[585]: Playlist is empty. Sleeping for 5 seconds
Jan 10 15:42:14 PiTVGISAND115 python[585]: Generating asset-list...
Jan 10 15:42:14 PiTVGISAND115 python[585]: Playlist is empty. Sleeping for 5 seconds
Jan 10 15:42:19 PiTVGISAND115 python[585]: Generating asset-list...
Jan 10 15:42:19 PiTVGISAND115 python[585]: Playlist is empty. Sleeping for 5 seconds

[image]

rusko124, 2019-01-10:
Hi @TheCazz 
Try to set the end time of the asset, you have gone beyond currently.

TheCazz, 2019-01-10:
Ahh it was so simple :)
Thanks. I think it maybe was that problem before too but I have to see when I get to the other place.

vpetersson, 2019-02-21:
Sounds like this issue has been resolved.

closed 2019-02-21 by vpetersson