Screenly/Anthias #1004: "Unable to run tvservice" logged in system info, nothing showing up on screen

Cryjack, 2019-01-03:
#### Overview of the Issue

I've recently updated two of our three Screenly's faultly with the experimental branch instead of the development branch via the update script. After updating back to development something is broken and the two are not showing up a picture anymore.
Systeminfo shows the attached error, while 'tvservice -s' works well on commandline.

#### Reproduction Steps

* Update to "Experimental" via Update Script
* Update back to "Development" via Update Script
* Showing error with tvservice

### Environment

* Raspberry Pi Hardware Version: 3
* Screenly OSE Version: latest dev

[image]

vpetersson, 2019-01-04:
@Cryjack This is a known issue. Essentially, it's related to #987. Since we have moved the services to run inside of Docker, we are unable to execute services on the host. Hence, we need a separate command executer (Celery in this case).

Cryjack, 2019-01-04:
Hi @vpetersson 

thank you for your response. So it seems that there is actually no way to change back to the non docker version without a complete reinstall of the raspberry?

There are 1200km between me and the Raspberry... :-o

vpetersson, 2019-01-04:
> So it seems that there is actually no way to change back to the non docker version without a complete reinstall of the raspberry?

Assuming you have SSH access, you can. All you need to do is to re-run the upgrade script and switch to the development or production version.

Cryjack, 2019-01-04:
> > So it seems that there is actually no way to change back to the non docker version without a complete reinstall of the raspberry?
> 
> Assuming you have SSH access, you can. All you need to do is to re-run the upgrade script and switch to the development or production version.

This is what I have done already a few times. It seems that some requirements are not given, even if the update script runs just fine without errors.

vpetersson, 2019-01-04:
That's strange. Could you post the logs?

Cryjack, 2019-01-04:
Yes, no problem. Is there a special screenly log that you need, or a debug mode? I can't find something screenly-related in /var/log/

Cryjack, 2019-01-04:
Got something like this from "journalctl"

```
Jan 04 11:55:49 screen3 containerd[460]: time="2019-01-04T11:55:49.765684226+01:00" level=info msg="shim containerd-shim started" address="/containerd
Jan 04 11:55:49 screen3 containerd[460]: time="2019-01-04T11:55:49.947582596+01:00" level=warning msg="OOM monitoring failed" error="cgroups: memory c
Jan 04 11:55:50 screen3 containerd[460]: time="2019-01-04T11:55:50.346284173+01:00" level=info msg="shim reaped" id=7a551685c9ce4a01682b01683321657388
Jan 04 11:55:50 screen3 dockerd[529]: time="2019-01-04T11:55:50.356744374+01:00" level=info msg="ignoring event" module=libcontainerd namespace=moby t
```

Cryjack, 2019-01-04:
Not sure if this is related, but the update script always shows this error (but even before the problem appears):

```
TASK [screenly : Run database migration] *************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": true, "cmd": ["python", "/home/pi/screenly/bin/migrate.py"], "delta": "0:00:00.158265", "end": "2019-01-04 12:08:31.681253", "msg": "non-zero return code", "rc": 1, "start": "2019-01-04 12:08:31.522988", "stderr": "Traceback (most recent call last):\n  File \"/home/pi/screenly/bin/migrate.py\", line 170, in <module>\n    migrate_drop_filename()\n  File \"/home/pi/screenly/bin/migrate.py\", line 159, in migrate_drop_filename\n    with open_db_get_cursor() as (cursor, _):\n  File \"/usr/lib/python2.7/contextlib.py\", line 17, in __enter__\n    return self.gen.next()\n  File \"/home/pi/screenly/bin/migrate.py\", line 53, in open_db_get_cursor\n    with sqlite3.connect(database, detect_types=sqlite3.PARSE_DECLTYPES) as conn:\nsqlite3.OperationalError: unable to open database file", "stderr_lines": ["Traceback (most recent call last):", "  File \"/home/pi/screenly/bin/migrate.py\", line 170, in <module>", "    migrate_drop_filename()", "  File \"/home/pi/screenly/bin/migrate.py\", line 159, in migrate_drop_filename", "    with open_db_get_cursor() as (cursor, _):", "  File \"/usr/lib/python2.7/contextlib.py\", line 17, in __enter__", "    return self.gen.next()", "  File \"/home/pi/screenly/bin/migrate.py\", line 53, in open_db_get_cursor", "    with sqlite3.connect(database, detect_types=sqlite3.PARSE_DECLTYPES) as conn:", "sqlite3.OperationalError: unable to open database file"], "stdout": "", "stdout_lines": []}
        to retry, use: --limit @/home/pi/screenly/ansible/site.retry
```

vpetersson, 2019-01-04:
Yes, that's highly relevant. @rusko124 probably made some backward incompatible changes to the database.

Cryjack, 2019-01-04:
Got it working.... OMG!

I'd always done the updates with sudo rights. That was why it doesn't find the path to the DB!
Doing it this way, it never comes to the dependency updates after the migration error. It just stops there and never proceeds the nessecary installs without any error than the migration.

Thank you @vpetersson for your help pointing me to the right direction.

closed 2019-01-04 by Cryjack

vpetersson, 2019-01-04:
Glad you got it working.

vpetersson, 2019-05-07:
@qbicdesign that shouldn't impact anything however.

qbicdesign, 2019-05-07:
It does, my system info page is empty, just like the screenshot in the first post.

vpetersson, 2019-05-08:
@qbicdesign Interesting. @rusko124 are you able to reproduce this?

qbicdesign, 2019-05-28:
I have since upgraded to experimental on 2 other raspberrys and the same thing happens, system info is blank apart from the table.

qbicdesign, 2019-05-28:
maybe we should open a separate issue for this?

vpetersson, 2019-05-28:
@qbicdesign yes please open a separate issue and provide steps for reproducing it per the template. Thanks.