Screenly/Anthias #1006: It gives me a 502 bad gateway error.

menarcarlos, 2019-01-06:
Hi folks, following the tutorial: https://www.youtube.com/watch?v=-P_OZRBZN74
I Have a issue to open the url: Error 502 Bad Gateway

rusko124, 2019-01-07:
Hi @menarcarlos !
Could you provide more information or write your actions step by step, please?

menarcarlos, 2019-01-07:
Hi rusko.

I'm using raspberry pi3 b+ (ARM7), I'm using balena.io to upload the project.

The steps followed to develop the application were taken from: https://www.balena.io/docs/learn/getting-started/raspberrypi3/python/

I using microSDXC 10 class formated with Etcher over Windows 10.

rusko124, 2019-01-08:
@menarcarlos 
`git push balena master` passes without error?
Then you should see something like this.
[image]
Have you reached this point?

ealmonte32, 2019-01-10:
@menarcarlos could you pls upload a screenshot of the bad gateway error exactly as you see it?

menarcarlos, 2019-01-11:
Hi to all.
I start to clone the repo :  git clone https://github.com/Screenly/screenly-ose.git

then...
c:\>balena login 		( Success )
c:\>balena push screenlyose 	 ( Success )
Unicorn printed			 ( Success )
Balena cloud (Updating)          ( Success )
https://5e3e0dc237e046cbb3359ff2ffbf8b5f.balena-devices.com/       - Unsuccess

So i went to the source code of the project and i found in the code of:
screenly-ose/settings.py

This lines:

`PORT = int(getenv('PORT', 8080))`
`LISTEN = getenv('LISTEN', '127.0.0.1')`

ealmonte32, 2019-01-14:
> Hi to all.
> I start to clone the repo : git clone https://github.com/Screenly/screenly-ose.git
>
> then...
> c:>balena login ( Success )
> c:>balena push screenlyose ( Success )
> Unicorn printed	( Success ) [...]

@menarcarlos Do you have physical access to the raspberry pi device? If so, can you pls paste the outcome of the following command:

sudo netstat -pln | grep :80

Thanks.

vpetersson, 2019-02-21:
Closing out issue due to lack of response.

closed 2019-02-21 by vpetersson